tetele/onju-voice-satellite #7: Wake Word sound

smoldersonline, 2024-03-12:
Thank you very much for sharing your work here!

I am new to ESPHome configuration, living and learning and picking up bits and pieces here and there.

I have successfully saved your latest configuration to my Onju Voice devices.

A thing that is still missing for my personal situation, is the device playing a sound when it detects the Wake Word. On Discord, user BigBobba was so kind to share [his code](https://discord.com/channels/429907082951524364/1211899225915265064).

I (inexperienced) pasted BigBobba's code (from Discord) into your (most recent) configuration. It works, but (as far as I can tell) there is one drawback. The template switch (from your onju-voice.yaml, line 537) to turn on/off the use of the Wake Word stops working - i.e. it will only very briefly turn off the Wake Word.

Both 1) the Wake Word sound, as well as 2) the template switch, are somewhat important to me. With trail and error I tried to fix it, but (no surprise) no luck ...

Your kind support would be greatly appreciated.

smoldersonline, 2024-03-12:
I apologise for being vague (I'm native Dutch).

This is my config file:
[onju-voice03 copy.yaml.zip](https://github.com/tetele/onju-voice-satellite/files/14575070/onju-voice03.copy.yaml.zip)

Basically your configuration, with (only) changes to the voice assistant section.

With this (changed) configuration, the device will play audio (How can I help) when the Wake Word  is detected. After that it functions as normal. So far, so good.

However, as a result of my changes to the original configuration file, the template switch (your onju-voice.yaml, line 537), will no longer properly work. It will only very briefly turn off the use of the Wake Work. After a split second, it will turn on again. I think my changes to the voice assistant section are triggering this behaviour.

I sincerely hope this is clear. If not, I will understand that you want to give up on my querie.

tetele, 2024-03-12:
OK, i can see how your additions are messing with the original behavior, but I'm not sure exactly how to fix it and can't test right now.

Try this (obviously, leave everything else in place):

```yaml
voice_assistant:
  on_tts_end:
    - media_player.play_media: !lambda return x;
    - light.turn_on:
        id: top_led
        blue: 0%
        red: 20%
        green: 100%
        effect: speaking
  on_end:
    - delay: 100ms
    - wait_until:
        not:
          media_player.is_playing: onju_out
    - delay: 600ms
    - script.execute: turn_on_wake_word
```

smoldersonline, 2024-03-13:
Wow - this works! Thanks a lot.

Just for the sake completeness, I slightly changed your above code:

from
 `        effect: speaking`
to
`        effect: pulse`

Again, thank you very much for your patience and kind support.

tetele, 2024-03-13:
Glad to hear it works!

closed 2024-03-13 by tetele

fantnhu, 2024-03-16:
I have modified the code based on the zip file and the fix. 
The speaker does not sound, neither the startup tone nor the tts response is played.
I made some text senors to monitor the events:
on_wake_word_detected: --> runs, but does not play the startup tone (neither with HA service call nor with native ESP playback)
on_stt_end --> successfully returns the command
on_tts_start --> successfully returns the response as text
on_tts_end --> successfully returns the URL of the response file (.mp3), but is not played by the player.

So it understands the question, does the commands, but does not speak. 
Of course, the answer works with the original code on the speaker, but I don't want to have to look at the speaker every time I talk to it to see if it understands the question or not.  That's why it would be good for me to have a startup sound when listening.

These settings seem to run twice:
[media_player:059]
[media_player:066]

For the 3rd day now I can't get a media player to play a media file :D 

Any ideas?
```python
voice_assistant:
  id: va
  microphone: onju_microphone
  media_player: onju_out
  use_wake_word: true
  on_start:
    - light.turn_on:
        id: top_led
        blue: 0%
        red: 0%
        green: 0%
        brightness: 50%
        effect: none
  on_wake_word_detected:
    - voice_assistant.stop
    - lambda: id(va).set_use_wake_word(false);
    - delay: 300ms
    - homeassistant.service:
        service: media_player.play_media
        data:
          media_content_id: 'https://MYDOMAIN.hu/local/audio/va_listening.mp3'
          media_content_type: audio/mp3
          entity_id: media_player.cortana_ai_living_room
    - wait_until:
        not:
          media_player.is_playing: onju_out
    - delay: 300ms
    - voice_assistant.start_continuous
  on_listening:
    - light.turn_on:
        id: top_led
        blue: 100%
        red: 0%
        green: 0%
        brightness: 100%
        effect: pulse
  on_stt_end:
    - text_sensor.template.publish:
        id: va_command_txt
        state: !lambda return x;
  on_tts_start:
    - text_sensor.template.publish:
        id: va_response_txt
        state: !lambda return x;
  on_tts_end:
    - media_player.play_media: !lambda return x;
    - light.turn_on:
        id: top_led
        blue: 0%
        red: 20%
        green: 100%
        effect: speaking
    - text_sensor.template.publish:
        id: va_response_url
        state: !lambda return x;
  on_end:
    - delay: 100ms
    - wait_until:
        not:
          media_player.is_playing: onju_out
    - delay: 600ms
    - script.execute: turn_on_wake_word
    
  on_client_connected:
    - delay: 2s
    - lambda: id(va).set_use_wake_word(true);
    - delay: 200ms
    - voice_assistant.start_continuous
``` 

> [10:06:01][D][voice_assistant:523]: Event Type: 4
[10:06:01][D][voice_assistant:551]: Speech recognised as: "Mondj egy számot"
[10:06:01][D][text_sensor:064]: 'AI Command': Sending state 'Mondj egy számot'
[10:06:01][D][voice_assistant:523]: Event Type: 5
[10:06:01][D][voice_assistant:556]: Intent started
[10:06:02][D][voice_assistant:523]: Event Type: 6
[10:06:02][D][voice_assistant:523]: Event Type: 7
[10:06:02][D][voice_assistant:579]: Response: "17"
[10:06:02][D][text_sensor:064]: 'AI Response TEXT': Sending state '17'
[10:06:02][D][voice_assistant:523]: Event Type: 8
[10:06:02][D][voice_assistant:599]: Response URL: "https://MYDOMAIN.hu/api/tts_proxy/0716d9708d321ffb6a00818614779e779925365c_hu-hu_d9ec7e5d00_tts.piper.mp3"
[10:06:02][D][voice_assistant:416]: State changed from AWAITING_RESPONSE to STREAMING_RESPONSE
[10:06:02][D][voice_assistant:422]: Desired state set to STREAMING_RESPONSE
[10:06:02][D][media_player:059]: 'Cortana AI Living Room' - Setting
[10:06:02][D][media_player:066]:   Media URL: https://MYDOMAIN.hu/api/tts_proxy/0716d9708d321ffb6a00818614779e779925365c_hu-hu_d9ec7e5d00_tts.piper.mp3
[10:06:02][D][media_player:059]: 'Cortana AI Living Room' - Setting
[10:06:02][D][media_player:066]:   Media URL: https://MYDOMAIN.hu/api/tts_proxy/0716d9708d321ffb6a00818614779e779925365c_hu-hu_d9ec7e5d00_tts.piper.mp3
[10:06:02][D][light:036]: 'top_led' Setting:
[10:06:02][D][light:059]:   Red: 20%, Green: 100%, Blue: 0%
[10:06:02][D][light:109]:   Effect: 'speaking'
[10:06:02][D][text_sensor:064]: 'AI Response URL': Sending state 'https://MYDOMAIN.hu/api/tts_proxy/0716d9708d321ffb6a00818614779e779925365c_hu-hu_d9ec7e5d00_tts.piper.mp3'
[10:06:02][W][component:214]: Component i2s_audio.media_player took a long time for an operation (0.52 s).
[10:06:02][W][component:215]: Components should block for at most 20-30ms.
[10:06:03][W][component:214]: Component i2s_audio.media_player took a long time for an operation (0.50 s).
[10:06:03][W][component:215]: Components should block for at most 20-30ms.
[10:06:03][D][light:036]: 'top_led' Setting:
[10:06:03][D][light:051]:   Brightness: 60%
[10:06:03][D][light:059]:   Red: 100%, Green: 0%, Blue: 100%
[10:06:03][D][light:109]:   Effect: 'listening_ww'

tetele, 2024-03-16:
Don't use HA services to play media, use the internal `media_player.play_media` service

fantnhu, 2024-03-16:
> Don't use HA services to play media, use the internal `media_player.play_media` service
Unfortunately, it doesn't work. It does not play the startup sound and it does not play the answer. 
Any other suggestions?

```
voice_assistant:
  id: va
  microphone: onju_microphone
  media_player: onju_out
  use_wake_word: true
  on_start:
    - light.turn_on:
        id: top_led
        blue: 0%
        red: 0%
        green: 0%
        brightness: 50%
        effect: none
  on_wake_word_detected:
    - voice_assistant.stop
    - lambda: id(va).set_use_wake_word(false);
    - delay: 300ms
    - media_player.play_media:
        id: onju_out
        media_url: 'https://MYDOMAIN.hu/local/audio/va_listening.mp3'    
    - wait_until:
        not:
          media_player.is_playing: onju_out
    - delay: 300ms
    - voice_assistant.start_continuous
  on_listening:
    - light.turn_on:
        id: top_led
        blue: 100%
        red: 0%
        green: 0%
        brightness: 100%
        effect: pulse
  on_tts_end:
    - media_player.play_media: !lambda return x;
    - light.turn_on:
        id: top_led
        blue: 0%
        red: 20%
        green: 100%
        effect: speaking
  on_end:
    - delay: 100ms
    - wait_until:
        not:
          media_player.is_playing: onju_out
    - delay: 600ms
    - script.execute: turn_on_wake_word
    
  on_client_connected:
    - delay: 2s
    - lambda: id(va).set_use_wake_word(true);
    - delay: 200ms
    - voice_assistant.start_continuous
``` 

```python
[12:50:27][D][voice_assistant:681]: STT by VAD end
[12:50:27][D][voice_assistant:416]: State changed from STREAMING_MICROPHONE to STOP_MICROPHONE
[12:50:27][D][voice_assistant:422]: Desired state set to AWAITING_RESPONSE
[12:50:27][D][voice_assistant:416]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[12:50:27][D][voice_assistant:416]: State changed from STOPPING_MICROPHONE to AWAITING_RESPONSE
[12:50:30][D][voice_assistant:523]: Event Type: 4
[12:50:30][D][voice_assistant:551]: Speech recognised as: "álmot"
[12:50:30][D][voice_assistant:523]: Event Type: 5
[12:50:30][D][voice_assistant:556]: Intent started
[12:50:32][D][voice_assistant:523]: Event Type: 6
[12:50:32][D][voice_assistant:523]: Event Type: 7
[12:50:32][D][voice_assistant:579]: Response: "Ha az álmaid valósággá válnának, akkor a "Yeelight color5 - zuhany" lámpa minden éjjel szivárvánnyá változna a fürdőszobában."
[12:50:32][D][voice_assistant:523]: Event Type: 8
[12:50:32][D][voice_assistant:599]: Response URL: "https://MYDOMAIN.hu/api/tts_proxy/bedf943832c3ec4fc093fca5407541b8daee8655_hu-hu_d9ec7e5d00_tts.piper.mp3"
[12:50:32][D][voice_assistant:416]: State changed from AWAITING_RESPONSE to STREAMING_RESPONSE
[12:50:32][D][voice_assistant:422]: Desired state set to STREAMING_RESPONSE
[12:50:32][D][media_player:059]: 'Cortana AI Living Room' - Setting
[12:50:32][D][media_player:066]:   Media URL: https://MYDOMAIN.hu/api/tts_proxy/bedf943832c3ec4fc093fca5407541b8daee8655_hu-hu_d9ec7e5d00_tts.piper.mp3
[12:50:32][D][media_player:059]: 'Cortana AI Living Room' - Setting
[12:50:32][D][media_player:066]:   Media URL: https://MYDOMAIN.hu/api/tts_proxy/bedf943832c3ec4fc093fca5407541b8daee8655_hu-hu_d9ec7e5d00_tts.piper.mp3
[12:50:32][D][light:036]: 'top_led' Setting:
[12:50:32][D][light:059]:   Red: 20%, Green: 100%, Blue: 0%
[12:50:32][D][light:109]:   Effect: 'speaking'
[12:50:32][D][voice_assistant:523]: Event Type: 2
[12:50:32][D][voice_assistant:613]: Assist Pipeline ended
[12:50:32][W][component:214]: Component i2s_audio.media_player took a long time for an operation (0.52 s).
[12:50:32][W][component:215]: Components should block for at most 20-30ms.
[12:50:33][W][component:214]: Component i2s_audio.media_player took a long time for an operation (0.51 s).
[12:50:33][W][component:215]: Components should block for at most 20-30ms.
```

smoldersonline, 2024-03-16:
When using homeassistant.service, make sure the device is allowed to make Home Assistant service calls.

[image]

fantnhu, 2024-03-16:
> When using homeassistant.service, make sure the device is allowed to make Home Assistant service calls.

thank you, but i am aware of it. i also used m5stack echo, the service call worked perfectly. 
Here it doesn't work with service call or internal media player.
(it doesn't play the reply message either, not just the startup sound)
apart from that the original code works, but i would like feedback on when the device is listening.

tetele, 2024-03-16:
Unfortunately, i'm on vacation until March 24th, with no access to my computer or devices at home.

fantnhu, 2024-03-17:
Enjoy the rest :) i'm sure the project will do great things