tetele/onju-voice-satellite #10: ModuleNotFoundError: No module named pip._internal.cli

VivantSenior, 2024-03-15:
Hi,

I received the following error when tried to compile new firmware. Tried to clean build files and create new device in ESPHome but no luck. My config:
HA: 2024.3.1
Operating System: 12.1
ESPHome: 2024.2.2
Hardware: Raspberry Pi 4

Most of my config is the same as in [main branch with wake word](https://github.com/tetele/onju-voice-satellite/blob/main/esphome/onju-voice-microwakeword.yaml) (just added some wifi and ap config).

I was able to compile the firmware using [ESPHome CLI](https://esphome.io/guides/cli.html) on Windows machine.

Error on RPi 4:
```
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/root/.platformio/penv/.espidf-4.4.6/lib/python3.11/site-packages/pip/__main__.py", line 29, in <module>
    from pip._internal.cli.main import main as _main
ModuleNotFoundError: No module named 'pip._internal.cli'
CalledProcessError: Command '['/root/.platformio/penv/.espidf-4.4.6/bin/python', '-m', 'pip', 'list', '--format=json', '--disable-pip-version-check']' returned non-zero exit status 1.:
  File "/usr/local/lib/python3.11/dist-packages/platformio/builder/main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.6.0/SCons/Script/SConscript.py", line 609:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.6.0/SCons/Script/SConscript.py", line 279:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/data/cache/platformio/platforms/espressif32/builder/main.py", line 312:
    target_elf = env.BuildProgram()
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.6.0/SCons/Util/envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.11/dist-packages/platformio/builder/tools/piobuild.py", line 61:
    env.ProcessProgramDeps()
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.6.0/SCons/Util/envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.11/dist-packages/platformio/builder/tools/piobuild.py", line 121:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.6.0/SCons/Util/envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.11/dist-packages/platformio/builder/tools/piobuild.py", line 347:
    SConscript(env.GetFrameworkScript(name), exports="env")
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.6.0/SCons/Script/SConscript.py", line 673:
    return method(*args, **kw)
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.6.0/SCons/Script/SConscript.py", line 609:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.6.0/SCons/Script/SConscript.py", line 279:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/data/cache/platformio/platforms/espressif32/builder/frameworks/espidf.py", line 1233:
    install_python_deps()
  File "/data/cache/platformio/platforms/espressif32/builder/frameworks/espidf.py", line 1110:
    installed_packages = _get_installed_pip_packages(python_exe_path)
  File "/data/cache/platformio/platforms/espressif32/builder/frameworks/espidf.py", line 1078:
    pip_output = subprocess.check_output(
  File "/usr/lib/python3.11/subprocess.py", line 466:
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.11/subprocess.py", line 571:
    raise CalledProcessError(retcode, process.args,
```

tetele, 2024-03-16:
I've never seen that error. Can you compile any other config with ESPHome on the Pi?

VivantSenior, 2024-03-16:
Me neither. During my research I have found this issue [#9795](https://github.com/pypa/pip/issues/9795). Yes, I can compile all of my 20 devices (ESP32 and ESP8286).

VivantSenior, 2024-05-03:
This issue in no longer present on ESPHome 2024.4.2.

closed 2024-05-03 by VivantSenior