tetele/onju-voice-satellite #11: [Feature request] Ability to change volume in Home Assistant

VivantSenior, 2024-03-15:
Hi,

I'm missing a possibility to change volume of the speaker in HA (with and without wake word). Would it be possible to expose it to HA?

Best,
Viv

fantnhu, 2024-03-16:
I have a similar problem. If I change the volume on the device, it is out of sync with the volume of the HomeAssistant media player entity and vice versa

tetele, 2024-03-16:
For the microWakeWord config with `speaker`, volume control is not yet supported in ESPHome, nothing i can do about that.

For openWakeWord, it should work out of the box. If it doesn't, please reply with the ESPHome error log.

fantnhu, 2024-03-16:
> Az openWakeWord esetében a dobozból kiindulva működnie kell. Ha nem, kérjük, válaszoljon az ESPHome hibanaplóval.

Thank you for your quick reply. I use OpenWakeWord on 4 devices. Unfortunately, none of them have the volume in sync. 
I can adjust the volume on the box, it works up and down. 
However, when I want to adjust the volume of the media player under HA, the volume changes under HA, but not on the box.

tetele, 2024-03-16:
That might be qn ESPHome bug (again, out of my control). Try adjusting the volume in HA during the time when the `media_player` is playing something (like a response).

VivantSenior, 2024-03-16:
> For the microWakeWord config with `speaker`, volume control is not yet supported in ESPHome, nothing i can do about that.
> 
> For openWakeWord, it should work out of the box. If it doesn't, please reply with the ESPHome error log.

Thank you. I created [#12](https://github.com/tetele/onju-voice-satellite/issues/12) feature request and found [#2619](https://github.com/esphome/feature-requests/issues/2619) which #12 is depended upon.

jherby2k, 2024-04-11:
From what i can tell, the only issue with the openWakeWord config is that the Home Assistant UI shows 100% volume. i can set the slider to 80% and the speaker gets set to 80%, but if i refresh the page in HA, it shows 100% again. So the volume just isn't being reflected back to HA.

I'd love to fix this for you but i'm still trying to understand how this all ties together TBH.

tetele, 2024-04-12:
> I'd love to fix this for you but i'm still trying to understand how this all ties together TBH. Please just let me know if this is an ESPHome bug and i'll open it there.

As far as I remember, that was an ESPHome bug. You *can* set the volume of the `media_player` from HA, but only while it's playing back a response.

VivantSenior, 2024-05-11:
Closing this feature request as it has been implemented in #39.

tetele, 2024-06-03:
Root cause: https://github.com/esphome/issues/issues/5877

tetele, 2024-06-04:
Can you try adding this to the config and checking if it solves the issue?

```yaml
external_components:
  - source: github://pr#6859
    components: i2s_audio
    refresh: 0s
```

tetele, 2024-06-21:
Fixed in ESPHome 2024.6 https://github.com/esphome/esphome/pull/6859

closed 2024-06-21 by tetele