tetele/onju-voice-satellite #12: Volume change button doesn't work

VivantSenior, 2024-03-15:
Hi,

I'm trying to lower speaker volume (it's really loud) but nothing changes (also no visual feedback provided via the LEDs). Before the PCB transplant I checked and it worked. I have [wake word](https://github.com/tetele/onju-voice-satellite/blob/main/esphome/onju-voice-microwakeword.yaml) version.

Any ideas what could be wrong? Is something missing in the config?

Best,
Viv

tetele, 2024-03-16:
> Before the PCB transplant I checked and it worked.

Of course. Both the volume controls and the logic were on the original PCB.

> I have [wake word](https://github.com/tetele/onju-voice-satellite/blob/main/esphome/onju-voice-microwakeword.yaml) version.

That's the microWakeWord version which doesn't have volume control, as per the README. The reason is that the `speaker` component in ESPHome does not support volume control, so there is nothing i can do about it.

VivantSenior, 2024-03-16:
Thank you. I have found feature request in the ESPHome repository [#2619](https://github.com/esphome/feature-requests/issues/2619) regarding ability to adjust volume in the speaker component.

tetele, 2024-04-18:
Addressed with a major overhaul in #39, which is just a test for now

VivantSenior, 2024-05-11:
Closing this feature request as it has been implemented in #39.

closed 2024-05-11 by VivantSenior