tetele/onju-voice-satellite #14: play "on_listening" sound

fantnhu, 2024-03-16:
Hi! 
I want to play sound effects when the device is listening. 
I did it in the configuration, but the device freezes after listening. 
[https://esphome.io/components/media_player/index.html#media-player-play-media-action](url)
On M5Stack Echo I could solve this with HomeAssistant service.
Is there a way to do this?

tetele, 2024-03-16:
See #7 

You can't both listen and play on the same time, you need to temporarily stop listening in order to play a sound

s00500, 2024-03-24:
Hey @tetele,

I have been playing with my own microWakeWord version in the last days and made a simple sound playback work.

Basically like this:
```
micro_wake_word:
  model: okay_nabu
  on_wake_word_detected:
    - logger.log: Local wakeword detected
    - speaker.play:
        data: !lambda return beep;
    - voice_assistant.start:
        wake_word: !lambda return wake_word;
```

The beep is included using a included .h file and currently consists out of a few random bytes... I should probably still search for a better way to encode these... cant be to much data though as microWakeWord already uses a lot of memory...

```
esphome:
  name: "${name}"
  includes:
    - beep.h
```

In beep.h:
```
std::vector<uint8_t> beep = { 0x52, 0x49, 0x46, 0x46, 0x46, 0xD8, 0x00, 0x00, 0x57, 0x41, 0x56, 0x45, 0x66, 0x6D, 0x74, 0x20, 0x10, 0x00, 0x00, 0x00, 0x01, 0x00, 0x02, 0x00, 0x44, 0xAC, 0x00, 0x00, 0x10, 0xB1, 0x02, 0x00, 0x04, 0x00, 0x10, 0x00, 0x4C, 0x49, 0x53, 0x54, 0x1A, 0x00, 0x00, 0x00, 0x49, 0x4E, 0x46, 0x4F, 0x49, 0x53, 0x46, 0x54, 0x0E, 0x00, 0x00, 0x00, 0x4C, 0x61, 0x76, 0x66, 0x35, 0x39, 0x2E, 0x32, 0x37, 0x2E, 0x31, 0x30, 0x30, 0x00, 0x64, 0x61, 0x74, 0x61, 0x00, 0xD8, 0x52, 0x49, 0x46, 0x46, 0x46, 0xD8, 0x00, 0x00, 0x57, 0x41, 0x56, 0x45, 0x66, 0x6D, 0x74, 0x20, 0x10, 0x00, 0x00, 0x00, 0x01, 0x00, 0x02, 0x00, 0x44, 0xAC, 0x00, 0x00, 0x10, 0xB1, 0x02, 0x00, 0x04, 0x00, 0x10, 0x00, 0x4C, 0x49, 0x53, 0x54, 0x1A, 0x00, 0x00, 0x00, 0x49, 0x4E, 0x46, 0x4F, 0x49, 0x53, 0x46, 0x54, 0x0E, 0x00, 0x00, 0x00, 0x4C, 0x61, 0x76, 0x66, 0x35, 0x39, 0x2E, 0x32, 0x37, 0x2E, 0x31, 0x30, 0x30, 0x00, 0x64, 0x61, 0x74, 0x61, 0x00, 0xD8, 0x52, 0x49, 0x46, 0x46, 0x46, 0xD8, 0x00, 0x00, 0x57, 0x41, 0x56, 0x45, 0x66, 0x6D, 0x74, 0x20, 0x10, 0x00, 0x00, 0x00, 0x01, 0x00, 0x02, 0x00, 0x44, 0xAC, 0x00, 0x00, 0x10, 0xB1, 0x02, 0x00, 0x04, 0x00, 0x10, 0x00, 0x4C, 0x49, 0x53, 0x54, 0x1A, 0x00, 0x00, 0x00, 0x49, 0x4E, 0x46, 0x4F, 0x49, 0x53, 0x46, 0x54, 0x0E, 0x00, 0x00, 0x00, 0x4C, 0x61, 0x76, 0x66, 0x35, 0x39, 0x2E, 0x32, 0x37, 0x2E, 0x31, 0x30, 0x30, 0x00, 0x64, 0x61, 0x74, 0x61, 0x00, 0xD8, 0x52, 0x49, 0x46, 0x46, 0x46, 0xD8, 0x00, 0x00, 0x57, 0x41, 0x56, 0x45, 0x66, 0x6D, 0x74, 0x20, 0x10, 0x00, 0x00, 0x00, 0x01, 0x00, 0x02, 0x00, 0x44, 0xAC, 0x00, 0x00, 0x10, 0xB1, 0x02, 0x00, 0x04, 0x00, 0x10, 0x00, 0x4C, 0x49, 0x53, 0x54, 0x1A, 0x00, 0x00, 0x00, 0x49, 0x4E, 0x46, 0x4F, 0x49, 0x53, 0x46, 0x54, 0x0E, 0x00, 0x00, 0x00, 0x4C, 0x61, 0x76, 0x66, 0x35, 0x39, 0x2E, 0x32, 0x37, 0x2E, 0x31, 0x30, 0x30, 0x00, 0x64, 0x61, 0x74, 0x61, 0x00, 0xD8, 0x52, 0x49, 0x46, 0x46, 0x46, 0xD8, 0x00, 0x00, 0x57, 0x41, 0x56, 0x45, 0x66, 0x6D, 0x74, 0x20, 0x10, 0x00, 0x00, 0x00, 0x01, 0x00, 0x02, 0x00, 0x44, 0xAC, 0x00, 0x00, 0x10, 0xB1, 0x02, 0x00, 0x04, 0x00, 0x10, 0x00, 0x4C, 0x49, 0x53, 0x54, 0x1A, 0x00, 0x00, 0x00, 0x49, 0x4E, 0x46, 0x4F, 0x49, 0x53, 0x46, 0x54, 0x0E, 0x00, 0x00, 0x00, 0x4C, 0x61, 0x76, 0x66, 0x35, 0x39, 0x2E, 0x32, 0x37, 0x2E, 0x31, 0x30, 0x30, 0x00, 0x64, 0x61, 0x74, 0x61, 0x00, 0xD8 };
```

tetele, 2024-03-26:
@s00500 first of all, sorry for the delay, I was on vacation.

Second, thanks for this! This is a good way to go about it, although it does use up a lot of memory, indeed.

Have you also tried playing the beep directly from a HA URL?

s00500, 2024-03-26:
No... there is still some trickyness in this.... I would love to have the speaker properly support URls...

I really like the setup with the IDF, wakeword works AMAZING!

But I cant have TTS this way, nor can I have volume control... 

As for the mediaplayer version: TTS works fine if you disable  UseWakeWord short before you send the tts from ha...

tetele, 2024-08-07:
This has been implemented for the MWW config in #74. See readme for instructions

closed 2024-08-07 by tetele

s00500, 2024-08-08:
Cool stuff, I had some issues here though... will test the updated config after coming back next week

iamjackg, 2024-08-15:
For some reason, in my setup this chime always ends up playing *after* the response is done playing.

tetele, 2024-08-16:
@iamjackg please open a new bug report. You may be playing a remote https URL, which might cause the issue