tetele/onju-voice-satellite #15: [Feature request] Switch to disable speaker of the microWakeWord variant

VivantSenior, 2024-03-17:
Hello,

as for now (more info in [#12](https://github.com/tetele/onju-voice-satellite/issues/12) and [#2619](https://github.com/esphome/feature-requests/issues/2619)) there is no way to change volume of the speaker in [microWakeWord variant](https://github.com/tetele/onju-voice-satellite/blob/main/esphome/onju-voice-microwakeword.yaml) (which is really loud) would it be possible to add a switch to Home Assistant that will mute it?

Best,
Viv

tetele, 2024-03-26:
Sorry for the delay, I was on vacation for a week. Yes, there's a simple way, actually.

Add a `name: Mute` line [somewhere here](https://github.com/tetele/onju-voice-satellite/blob/427110b72a44cf84ecbf9e3e0a61ad9e148b4625/esphome/onju-voice-microwakeword.yaml#L495). That switch is only used internally at the moment, to enable the DAC when sound need to be played. You can expose the switch in HA by giving it a name.

However, the root cause will only be solved when the volume will be controlled for `speaker` components OR when we will be able to use a `media_player` in `esp-idf`. Until then, the microWakeWord config is considered experimental and I don't see practical reasons to implement a lot of workarounds. It's only meant as a proof of concept for now.

tetele, 2024-03-26:
Yes, but last time i tested it i couldn't get it to work on the Onju. I'll give it another go in the following week.

tetele, 2024-04-18:
I'm going to close this now that #39 is available (albeit not merged yet)

closed 2024-04-18 by tetele