tetele/onju-voice-satellite #18: Using as TTS device in Home Assistant

ripvega, 2024-03-25:
I've just set up my first one and it's working great as a voice assistant device.

What I can't seem to get is TTS working.

In services, it doesn't show up in the list of TTS devices.

What am I missing?

tetele, 2024-03-26:
Do you want to send non-conversation-related TTS announcements to the device? Which config are you using?

ripvega, 2024-03-26:
I just want to be able to do something like this, as per my old google version:
`
      - service: tts.cloud_say
        metadata: {}
        data:
          cache: true
          entity_id: media_player.garage_speaker
          message: A vehicle is arriving`

AFAIK, I'm just using the config from the video tutorial.

`substitutions:
  name: "onju-voice-kitchen"
  friendly_name: "Onju Voice kitchen"
  project_version: "1.0.0"
  device_description: "Onju Voice Satellite with ESPHome software"

esphome:
  name: '${name}'
  comment: '${device_description}'
  name_add_mac_suffix: true
  project:
    name: tetele.onju_voice_satellite
    version: '${project_version}'
  min_version: 2023.11.6
  on_boot:
    then:
      - light.turn_on:
          id: top_led
          effect: slow_pulse
          red: 100%
          green: 60%
          blue: 0%
      - wait_until:
          condition:
            wifi.connected:
      - light.turn_on:
          id: top_led
          effect: pulse
          red: 0%
          green: 100%
          blue: 0%
      - wait_until:
          condition:
            api.connected:
      - light.turn_on:
          id: top_led
          effect: none
          red: 0%
          green: 100%
          blue: 0%
      - delay: 1s
      - script.execute: reset_led

dashboard_import:
  package_import_url: github://tetele/onju-voice-satellite/esphome/onju-voice.yaml@main

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino

# Enable logging
logger:

# Allow OTA updates
ota:
  password: !secret OTA_password

# Allow provisioning Wi-Fi via serial
[... 467 lines cut ...]
          switch(button) {
            case 0:
              id(volume_down).set_threshold(newthresh);
              break;
            case 1:
              id(action).set_threshold(newthresh);
              break;
            case 2:
              id(volume_up).set_threshold(newthresh);
              break;
            default:
              ESP_LOGE("touch_calibration", "Invalid button ID (%d)", button);
              return;
          }

switch:
  - platform: template
    name: Use Wake Word
    id: use_wake_word
    optimistic: true
    restore_mode: RESTORE_DEFAULT_ON
    on_turn_on:
      - script.execute: turn_on_wake_word
    on_turn_off:
      - script.execute: turn_off_wake_word    `

tetele, 2024-03-26:
You can format code by placing it between triple backticks `` ``` some code ``` `` or by pressing the "code" button in the toolbar (`<>`)

That said, your service call should be working (it does for me on the same config), but i see that the config you pasted is for a device called `Onju Voice kitchen` and your service call is for `media_player.garage_speaker`. Are you sure you're triggering the correct device?

ripvega, 2024-03-26:
Thank you. Unfortunatley, for some reason the code formatting back ticks don't like Arc browser. I'd assumed that they worked. Apologies.

The service call is an example of what I want to use.

When I use dev tools/services, my onju devices are not available to the TTS service. They don't show up as media.player in dev tools either. Hence the problem.

[image]

They are working fine as Voice Assistants though.

tetele, 2024-03-26:
Ah! Are you using the microWakeWord config? If so, that does not expose a `media_player`

ripvega, 2024-03-26:
Not that I am aware of. How would I check that?

I do have wakeword turned on.

tetele, 2024-03-26:
Nevermind, I am an idiot. You config provides the answer: no, you're not using the MWW config.

Weird that the `media_player` is not exposed. Could you remove the device and add it again, please?

ripvega, 2024-03-26:
Sure. And thank you for your help.

All three of my Onjus are the same.

I will try removal tomorrow.

ripvega, 2024-03-26:
All three, on initial integration, had the media player named as media_player.none, media_player.none_1 and media_player.none_2.

I've renamed the entities and they now show up in services.

I've been trying this:

```service: tts.speak
data:
  cache: true
  media_player_entity_id: media_player.onju_kitchen
  message: hello
target:
  entity_id: media_player.onju_kitchen```

but no speech. The service thinks it completes but nothing shows in the logbook that it was activated.

TTS works with google speakers.

Volume is fine as it works as voice assistant.

Any ideas?

ripvega, 2024-03-27:
I've deleted and then re-added my kitchen speaker. Here's what I see now:
[image]
It's still not showing up as a media player.

As you can see, the media player icon shows up with the name 'none'.

Clicking on it gives this:

[image]

The volume buttons and mute don't work and still no TTS.

Does this give any clues as to the problem?

jherby2k, 2024-04-11:
why its called "none" - not sure. The config does set the name to "None" (look at the media_player section) but that should set the media player's name to use the friendly name. Are you maybe not setting the friendly_name? Anyway, the name is just cosmetic - it should still work.

The media player volume up/down controls don't seem to work, but the slider does set the volume (it just doesn't update the UI - not sure why).

If you turn off "Use Wake Word" you should be able to play stuff to the speaker. It currently can't listen and play at the same time.

ripvega, 2024-04-11:
Thank you. I will give that a try.

PR #29 (2024-04-12): Add friendly name to config
Although defined as a substitution, the `friendly_name` was missing from `esphome`.

[According to this](https://www.esphome.io/components/esphome#esphome-configuration-variables), the implications were concerning components defined with `name: None`

Should fix #18
closed 2024-04-12 by tetele