tetele/onju-voice-satellite #21: Device doesn't wake up after period of being idle

churchofnoise, 2024-03-29:
I have 3 former Nest Mini 2's with the PCB board and the config provided here.
I have however noticed that each of them fails to  'awake' after having been idle for an extended period of time (with the listening turned off). The only solution is to hard cycle the device.

Could you please look into this?
Thanks!

cibernox, 2024-06-07:
I confirm the same symptoms. The device responds to the wake work once when freshly booted, then goes to idle and it won't answer ever again.

These are the last lines of the logs. After becoming IDLE, it will never respond again:
```
[15:59:25][D][micro_wake_word:177]: State changed from IDLE to START_MICROPHONE
[15:59:25][D][micro_wake_word:115]: Starting Microphone
[15:59:25][D][esp_adf_pipeline:050]: Starting request, current state STOPPED
[15:59:25][D][esp_adf_pipeline:302]: State changed from STOPPED to PREPARING
[15:59:25][D][micro_wake_word:177]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[15:59:25][D][esp_adf_pipeline:302]: State changed from PREPARING to STARTING
[15:59:25][D][esp-idf:000]: I (38114) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:8400959 Bytes, Inter:161224 Bytes, Dram:161224 Bytes

[15:59:25][D][esp-idf:000][i2s_in]: I (38116) AUDIO_ELEMENT: [i2s_in] AEL_MSG_CMD_RESUME,state:1

[15:59:25][D][esp-idf:000]: I (38118) AUDIO_PIPELINE: Pipeline started

[15:59:25][I][esp_adf_pipeline:214]: [ pcm_reader ] status: 14
[15:59:25][I][esp_adf_pipeline:214]: [ i2s_in ] status: 14
[15:59:25][I][esp_adf_pipeline:214]: [ pcm_reader ] status: 12
[15:59:25][D][esp_adf_pipeline:131]: Check element [i2s_in] status, 3
[15:59:25][D][esp_adf_pipeline:131]: Check element [pcm_reader] status, 3
[15:59:25][D][esp_adf_pipeline:302]: State changed from STARTING to RUNNING
[15:59:25][D][micro_wake_word:177]: State changed from STARTING_MICROPHONE to DETECTING_WAKE_WORD
[15:59:25][I][esp_adf_pipeline:214]: [ i2s_in ] status: 12
[15:59:31][D][micro_wake_word:362]: Wake word sliding average probability is 0.545 and most recent probability is 0.976
[15:59:31][D][micro_wake_word:128]: Wake Word Detected
[15:59:31][D][micro_wake_word:177]: State changed from DETECTING_WAKE_WORD to STOP_MICROPHONE
[15:59:31][D][micro_wake_word:134]: Stopping Microphone
[15:59:31][D][esp_adf_pipeline:302]: State changed from RUNNING to STOPPING
[15:59:31][D][micro_wake_word:177]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[15:59:31][D][esp-idf:000][i2s_in]: W (43731) AUDIO_ELEMENT: OUT-[i2s_in] AEL_IO_ABORT

[15:59:31][D][esp-idf:000][i2s_in]: W (43733) AUDIO_ELEMENT: OUT-[i2s_in] AEL_IO_ABORT

[15:59:31][D][esp-idf:000][i2s_in]: W (43735) AUDIO_ELEMENT: OUT-[i2s_in] AEL_IO_ABORT

[15:59:31][D][esp-idf:000][i2s_in]: W (43738) AUDIO_ELEMENT: OUT-[i2s_in] AEL_IO_ABORT

[15:59:31][D][esp_adf_pipeline:302]: State changed from STOPPING to STOPPED
[15:59:31][D][micro_wake_word:177]: State changed from STOPPING_MICROPHONE to IDLE
```

tetele, 2024-06-07:
@cibernox that's a different issue and it's caused by this https://github.com/esphome/issues/issues/5794