tetele/onju-voice-satellite #22: No sound out of the speaker

outdooricon, 2024-04-05:
Hi, I absolutely love this project and all of the hard work that you and others have put into this. I have nothing respect and awe for you all. I have a question and was wondering if you who are much more knowledgable would know what to do.

I have two new onju voice Nest mini's that I've put together. I just got a batch from PCBWay and have installed in a brand new Nest Mini 2 as well as a few years old Nest Mini 2. Neither one will output sound. They respond to my wakeword, process my speech and I see sound files created which when I play on PC, sound great. However, the Mini doesn't want to play it. Also, it does an odd thing where it disconnects wifi right after it tries to play. 

```
[17:04:59][D][voice_assistant:532]: Wake word detected
[17:04:59][D][voice_assistant:523]: Event Type: 3
[17:04:59][D][voice_assistant:537]: STT started
[17:04:59][D][light:036]: 'top_led' Setting:
[17:04:59][D][light:051]:   Brightness: 100%
[17:04:59][D][light:059]:   Red: 100%, Green: 100%, Blue: 100%
[17:04:59][D][light:109]:   Effect: 'listening'
[17:04:59][D][voice_assistant:523]: Event Type: 11
[17:04:59][D][voice_assistant:677]: Starting STT by VAD
[17:05:00][D][voice_assistant:523]: Event Type: 12
[17:05:00][D][voice_assistant:681]: STT by VAD end
[17:05:00][D][voice_assistant:416]: State changed from STREAMING_MICROPHONE to STOP_MICROPHONE
[17:05:00][D][voice_assistant:422]: Desired state set to AWAITING_RESPONSE
[17:05:00][D][voice_assistant:416]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[17:05:00][D][light:036]: 'top_led' Setting:
[17:05:00][D][light:051]:   Brightness: 70%
[17:05:00][D][light:059]:   Red: 0%, Green: 20%, Blue: 100%
[17:05:00][D][light:109]:   Effect: 'processing'
[17:05:00][D][voice_assistant:416]: State changed from STOPPING_MICROPHONE to AWAITING_RESPONSE
[17:05:02][D][voice_assistant:523]: Event Type: 4
[17:05:02][D][voice_assistant:551]: Speech recognised as: " What time is it?"
[17:05:02][D][voice_assistant:523]: Event Type: 5
[17:05:02][D][voice_assistant:556]: Intent started
[17:05:02][D][voice_assistant:523]: Event Type: 6
[17:05:02][D][voice_assistant:523]: Event Type: 7
[17:05:02][D][voice_assistant:579]: Response: "Sorry, I couldn't understand that"
[17:05:02][D][voice_assistant:523]: Event Type: 8
[17:05:02][D][voice_assistant:599]: Response URL: "http://192.168.1.32:8123/api/tts_proxy/dae2cdcb27a1d1c3b07ba2c7db91480f9d4bfd8f_en-gb_183ef190f5_tts.piper.mp3"
[17:05:02][D][voice_assistant:416]: State changed from AWAITING_RESPONSE to STREAMING_RESPONSE
[17:05:02][D][voice_assistant:422]: Desired state set to STREAMING_RESPONSE
[17:05:02][D][media_player:059]: 'Onju Voice v3 dde0c8 - #2' - Setting
[17:05:02][D][media_player:066]:   Media URL: http://192.168.1.32:8123/api/tts_proxy/dae2cdcb27a1d1c3b07ba2c7db91480f9d4bfd8f_en-gb_183ef190f5_tts.piper.mp3
[17:05:02][D][media_player:059]: 'Onju Voice v3 dde0c8 - #2' - Setting
[17:05:02][D][media_player:066]:   Media URL: http://192.168.1.32:8123/api/tts_proxy/dae2cdcb27a1d1c3b07ba2c7db91480f9d4bfd8f_en-gb_183ef190f5_tts.piper.mp3
[17:05:02][D][light:036]: 'top_led' Setting:
[17:05:02][D][light:059]:   Red: 20%, Green: 100%, Blue: 0%
[17:05:02][D][light:109]:   Effect: 'speaking'
[17:05:02][D][voice_assistant:523]: Event Type: 2
[17:05:02][D][voice_assistant:613]: Assist Pipeline ended
WARNING onju-voice-v3-2-dde0c8 @ 192.168.20.57: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for onju-voice-v3-2-dde0c8 @ 192.168.20.57
WARNING Disconnected from API
INFO Successfully connected to onju-voice-v3-2-dde0c8 @ 192.168.20.57 in 0.004s
INFO Successful handshake with onju-voice-v3-2-dde0c8 @ 192.168.20.57 in 0.094s
[17:05:30][D][voice_assistant:523]: Event Type: 0
[17:05:30][D][voice_assistant:523]: Event Type: 2
[17:05:30][D][voice_assistant:613]: Assist Pipeline ended
[17:05:30][D][voice_assistant:416]: State changed from STREAMING_MICROPHONE to IDLE
[17:05:30][D][voice_assistant:422]: Desired state set to IDLE
[17:05:30][D][voice_assistant:416]: State changed from IDLE to START_PIPELINE
[17:05:30][D][voice_assistant:422]: Desired state set to START_MICROPHONE
[17:05:30][D][voice_assistant:202]: Requesting start...
[17:05:30][D][voice_assistant:416]: State changed from START_PIPELINE to STARTING_PIPELINE
[17:05:30][D][voice_assistant:437]: Client started, streaming microphone
[17:05:30][D][voice_assistant:416]: State changed from STARTING_PIPELINE to STREAMING_MICROPHONE
[17:05:30][D][voice_assistant:422]: Desired state set to STREAMING_MICROPHONE
[17:05:30][D][voice_assistant:523]: Event Type: 1
[17:05:30][D][voice_assistant:526]: Assist Pipeline running
[17:05:30][D][voice_assistant:523]: Event Type: 9
[17:05:30][D][light:036]: 'top_led' Setting:
[17:05:30][D][light:051]:   Brightness: 60%
[17:05:30][D][light:059]:   Red: 100%, Green: 0%, Blue: 100%
[17:05:30][D][light:085]:   Transition length: 0.1s
```

What the log doesn't show is that it starts flashing green after it does a couple of orange fade-outs. Is there anything that I'm doing wrong? Do I need to consider that my whole batch of pcb's are bad? I don't see anything physically obvious wrong with the boards.

Thank you for any help you can provide

tetele, 2024-04-06:
The green flashing means (as shown in the logs, too) that the connection to HA has been reset for some reason. In my experience, that is usually due to networking issues, which depend on your particular setup.

How are you running HA? HAOS? Is the server in the same network as the Onju? Do you have any firewall in place?

outdooricon, 2024-04-06:
That was the help I needed! I deep dove into my firewall settings and found some issues. Fixed and now sound is all better. It works great, thank you!

closed 2024-04-06 by outdooricon