tetele/onju-voice-satellite #23: Purple - 4 lights

nigels0, 2024-04-07:


nigels0, 2024-04-08:
Well it isn't blinking - just a solid four sets of purple lights.

On powering on, it goes white, then flashes green for a while then just the four purple lights. I've connected it to HA (it seems OK) - here is the log:
<
INFO ESPHome 2024.3.2
INFO Reading configuration /config/esphome/onju-1.yaml...
INFO Starting log output from 192.168.10.212 using esphome API
INFO Successfully connected to onju-voice1 @ 192.168.10.212 in 0.006s
INFO Successful handshake with onju-voice1 @ 192.168.10.212 in 0.077s
[16:52:28][I][app:102]: ESPHome version 2024.3.2 compiled on Apr  8 2024, 16:44:37
[16:52:28][C][wifi:580]: WiFi:
[16:52:28][C][wifi:408]:   Local MAC: 80:65:99:C6:44:70
[16:52:28][C][wifi:413]:   SSID: 'WaysIoT'[redacted]
[16:52:28][C][wifi:416]:   IP Address: 192.168.10.212
[16:52:28][C][wifi:420]:   BSSID: 26:5A:4C:2C:EE:D5[redacted]
[16:52:28][C][wifi:421]:   Hostname: 'onju-voice1'
[16:52:28][C][wifi:423]:   Signal strength: -62 dB ▂▄▆█
[16:52:28][C][wifi:427]:   Channel: 6
[16:52:28][C][wifi:428]:   Subnet: 255.255.255.0
[16:52:28][C][wifi:429]:   Gateway: 192.168.10.1
[16:52:28][C][wifi:430]:   DNS1: 1.1.1.1
[16:52:28][C][wifi:431]:   DNS2: 9.9.9.9
[16:52:28][C][logger:166]: Logger:
[16:52:28][C][logger:167]:   Level: DEBUG
[16:52:28][C][logger:169]:   Log Baud Rate: 115200
[16:52:28][C][logger:170]:   Hardware UART: USB_CDC
[16:52:28][C][template.number:050]: Template Number 'Touch threshold percentage'
[16:52:28][C][template.number:051]:   Optimistic: YES
[16:52:28][C][template.number:052]:   Update Interval: never
[16:52:28][C][esp32_rmt_led_strip:175]: ESP32 RMT LED Strip:
[16:52:28][C][esp32_rmt_led_strip:176]:   Pin: 11
[16:52:28][C][esp32_rmt_led_strip:177]:   Channel: 0
[16:52:28][C][esp32_rmt_led_strip:202]:   RGB Order: GRB
[16:52:28][C][esp32_rmt_led_strip:203]:   Max refresh rate: 0
[16:52:28][C][esp32_rmt_led_strip:204]:   Number of LEDs: 6
[16:52:28][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Disable wake word'
[16:52:28][C][gpio.binary_sensor:016]:   Pin: GPIO38
[16:52:28][C][light:103]: Light 'leds'
[16:52:28][C][light:105]:   Default Transition Length: 0.0s
[16:52:28][C][light:106]:   Gamma Correct: 2.80
[16:52:28][C][light:103]: Light 'left_led'
[16:52:28][C][light:105]:   Default Transition Length: 1.0s
[16:52:28][C][light:106]:   Gamma Correct: 2.80
[16:52:28][C][light:103]: Light 'top_led'
[16:52:28][C][light:105]:   Default Transition Length: 1.0s
[16:52:28][C][light:106]:   Gamma Correct: 2.80
[16:52:28][C][light:103]: Light 'right_led'
[16:52:28][C][light:105]:   Default Transition Length: 1.0s
[16:52:28][C][light:106]:   Gamma Correct: 2.80
[16:52:28][C][template.switch:068]: Template Switch 'Use Wake Word'
[16:52:28][C][template.switch:091]:   Restore Mode: restore defaults to ON
[16:52:28][C][template.switch:057]:   Optimistic: YES
[16:52:28][C][esp32_touch:073]: Config for ESP32 Touch Hub:
[16:52:28][C][esp32_touch:074]:   Meas cycle: 0.80ms
[16:52:28][C][esp32_touch:075]:   Sleep cycle: 2.00ms
[16:52:28][C][esp32_touch:095]:   Low Voltage Reference: 0.8V
[16:52:28][C][esp32_touch:115]:   High Voltage Reference: 2.4V
[16:52:28][C][esp32_touch:135]:   Voltage Attenuation: 0V
[16:52:28][C][esp32_touch:169]:   Filter mode: IIR_16
[16:52:28][C][esp32_touch:170]:   Debounce count: 2
[16:52:28][C][esp32_touch:171]:   Noise threshold coefficient: 0
[16:52:28][C][esp32_touch:172]:   Jitter filter step size: 0
[16:52:28][C][esp32_touch:191]:   Smooth level: IIR_2
[16:52:28][C][esp32_touch:213]:   Denoise grade: BIT8
[16:52:28][C][esp32_touch:245]:   Denoise capacitance level: L0
[16:52:28][C][esp32_touch:260]:   Touch Pad 'volume_down'
[16:52:28][C][esp32_touch:261]:     Pad: T4
[16:52:28][C][esp32_touch:262]:     Threshold: 483444
[16:52:28][C][status:034]: Status Binary Sensor 'api_connection'
[16:52:28][C][status:034]:   Device Class: 'connectivity'
[16:52:28][C][captive_portal:088]: Captive Portal:
[16:52:28][C][mdns:115]: mDNS:
[16:52:28][C][mdns:116]:   Hostname: onju-voice1
[16:52:28][C][ota:096]: Over-The-Air Updates:
[16:52:28][C][ota:097]:   Address: onju-voice1.local:3232
[16:52:28][C][ota:103]:   OTA version: 2.
[16:52:28][C][api:139]: API Server:
[16:52:28][C][api:140]:   Address: onju-voice1.local:6053
[16:52:28][C][api:142]:   Using noise encryption: YES
[16:52:28][C][audio:203]: Audio:
[16:52:28][C][audio:225]:   External DAC channels: 1
[16:52:28][C][audio:226]:   I2S DOUT Pin: 12
[16:52:28][C][audio:227]:   Mute Pin: GPIO21
>

nigels0, 2024-04-10:
**definitely purple! 

Sorry - I wasn't clear - yes the buttons work - volume up gives:**

[15:05:11][D][binary_sensor:036]: 'volume_down': Sending state ON
[15:05:11][D][light:036]: 'left_led' Setting:
[15:05:11][D][light:047]:   State: ON
[15:05:11][D][light:085]:   Transition length: 1.0s
[15:05:11][D][media_player:059]: 'Onju Voice 1' - Setting
[15:05:11][D][media_player:069]:   Volume: 0.85
[15:05:11][D][light:036]: 'top_led' Setting:
[15:05:11][D][light:109]:   Effect: 'show_volume'
[15:05:11][D][binary_sensor:036]: 'volume_down': Sending state OFF
[15:05:11][D][light:036]: 'left_led' Setting:
[15:05:11][D][light:047]:   State: OFF
[15:05:11][D][light:085]:   Transition length: 1.0s
[15:05:12][D][light:036]: 'top_led' Setting:
[15:05:12][D][light:051]:   Brightness: 100%
[15:05:12][D][light:059]:   Red: 100%, Green: 0%, Blue: 100%
[15:05:12][D][light:085]:   Transition length: 1.0s
[15:05:12][D][light:091]:   Effect: 'None'

**Here's my config file:**

captive_portal:

substitutions:
  name: "onju-voice1"
  friendly_name: "Onju Voice 1"
  

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: false
  min_version: 2023.10.1
  on_boot:
    then:
      - light.turn_on:
          id: top_led
          effect: slow_pulse
          red: 100%
          green: 60%
          blue: 0%
      - wait_until:
          condition:
            wifi.connected:
      - light.turn_on:
          id: top_led
          effect: pulse
          red: 0%
          green: 100%
          blue: 0%
      - wait_until:
          condition:
            api.connected:
      - light.turn_on:
          id: top_led
          effect: none
          red: 0%
          green: 100%
          blue: 0%
      - delay: 1s
      - script.execute: reset_led

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino

[... 454 lines cut ...]
          switch(button) {
            case 0:
              id(volume_down).set_threshold(newthresh);
              break;
            case 1:
              id(action).set_threshold(newthresh);
              break;
            case 2:
              id(volume_up).set_threshold(newthresh);
              break;
            default:
              ESP_LOGE("touch_calibration", "Invalid button ID (%d)", button);
              return;
          }

switch:
  - platform: template
    name: Use Wake Word
    id: use_wake_word
    optimistic: true
    restore_mode: RESTORE_DEFAULT_ON
    on_turn_on:
      - script.execute: turn_on_wake_word
    on_turn_off:
      - script.execute: turn_off_wake_word

tetele, 2024-04-10:
Please [format the code](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#quoting-code) properly.

nigels0, 2024-04-10:
Sorry. Done.

tetele, 2024-04-11:
Ah, now I get it! I suggest you either update to the latest version of [my config](https://github.com/tetele/onju-voice-satellite/blob/main/esphome/onju-voice.yaml) or request support from the guy from where you got your config :)

Unlike Mark, I actually keep it up to date, because I understand what's there since I've written it myself.

nigels0, 2024-04-11:
yes, that fixed it  thanks.

closed 2024-04-11 by nigels0