tetele/onju-voice-satellite #24: Onju Voice freezes after speaker output

don86nl, 2024-04-07:
Hi,

I am using your config but cannot get my device to work. Everything seems to be installed correctly into ESPHome and Home Assistant. I
However, when the output it supposed to happen (a voice response, tts or play media action) the device freezes and reboots without any output...

Any idea? I am using the non-localwakeword version of your satelite. 
Thanks in advance!

tetele, 2024-04-08:
Can you please provide the ESPHome logs prior to the reboot happening? Also, what version of ESPHome and HA are you using?

don86nl, 2024-04-08:
Woops, my bad. While checking the logs I noticed an error in my wifi setup. 
Thanks for the quick response, @tetele :)