tetele/onju-voice-satellite #35: Volume control from Home Assistant

ripvega, 2024-04-16:
### Flavor

MicroWakeWord

### Checklist

- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/tetele/onju-voice-satellite/issues?q=is%3Aissue+label%3A%22bug%22+)..
- [X] I am able to consult debug logging for my ESPHome instance.

### Describe the issue

I don't seem to be able to make volume changes ether from service calls or using dev tools.

How should this be done? The volume seems a good bit lower than the original Google.

### Reproduction steps

I have tried changing volume_level: but it doesn't seem to affect the volume.

Over what range should volume_level: be set?

As [per my last issue](https://github.com/tetele/onju-voice-satellite/issues/18#issue-2206766399), the volume controls on the UI have no effect either.

tetele, 2024-04-17:
If you're using MicroWakeWord, there are no volume controls, as the output is a `speaker` component, not a `media_player`.

If, however, that was a mistake and you're using a `media_player`, then both the side touch controls work and you can also set the volume from HA, but only while the speaker is playing something (that's a bug in ESPHome).

ripvega, 2024-04-17:
Apologies. Yes, I'm not using MicroWakeWord.

I've used the touch controls to max and it's still not very loud.

I've been trying variants of the code below, without them passing the yaml code check. It doesn't like `volume_level: `as an option.

`- service: tts.speak
target:
  entity_id: tts.home_assistant_cloud
data:
  cache: true
  message: well "{{ states('sensor.red_light_therapy_mode') }}" mode
  media_player_entity_id: media_player.onju_spa
  options:
    voice: RyanNeural
  volume_level: 1`

tetele, 2024-06-03:
Likely cause: https://github.com/esphome/issues/issues/5877

tetele, 2024-06-04:
This will probably not fix the overall low volume issue, but can you try adding this to the config and checking if it solves the volume control issue?

```yaml
external_components:
  - source: github://pr#6859
    components: i2s_audio
    refresh: 0s
```

tetele, 2024-06-21:
Fixed in ESPHome 2024.6 https://github.com/esphome/esphome/pull/6859

closed 2024-06-21 by tetele