tetele/onju-voice-satellite #41: Long tap to toggle wakeword detection

vhsdream, 2024-04-18:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/tetele/onju-voice-satellite/issues?q=is%3Aissue+label%3A%22feature-request%22+).

### Is your feature request related to a problem? Please describe.

I don't always want the device listening for a wakeword as sometimes there are false positives - for example when I am watching a show or movie, or if I'm having an important call. There is an existing switch exposed in Home Assistant of course, and that works, but sometimes I want to be able to toggle detection on the device itself.

### Describe the solution you'd like

I would like to be able to do a longer tap on the onju which will toggle wakeword detection, and have there be visual feedback of the toggle coming from the LEDs.

### Describe alternatives you've considered

I have a semi-working version of this in a fork, however the visual feedback when the toggle disables detection does not seem to work properly (the LEDs all flash red). I think it is something wrong with the way I have configured it. If this sounds like a good feature that people want, I can make a PR.

### Additional context

Are all of these fields really required? Anyway, I was just going to make a PR but wanted to check if this wasn't too much 'feature creep', and if others had recommendations for further enhancements to the feature request.

jherby2k, 2024-04-18:
What's wrong with the physical mute switch already built in?

vhsdream, 2024-04-18:
Nothing the matter with it really - I just find that I never use it. Doing a brief touch on the top is far more convenient :)

tetele, 2024-04-19:
You already have two methods of toggling the wake word - one physical and one virtual. They both need to be on for wake word to be running.

You can automate on the virtual one - "when a movie starts, turn off wake word". You can also expose it and do "hey Nabu, turn off use wake word" or another custom command. You can also tap the top and say your custom command (turn on/off the wake word.

I feel like long tapping for this purpose would be a bit of an overkill.

vhsdream, 2024-04-19:
All good points! I guess it comes down to personal preference, as I have found that I never use the physical mute switch, and haven't gotten around to automating the toggling that much yet. For now I will keep using the long tap in my fork since I like it :)

closed 2024-04-19 by tetele