tetele/onju-voice-satellite #42: Volume change on device no visual response

Mugga6315, 2024-05-02:
### Flavor

OpenWakeWord or no wake word

### Checklist

- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/tetele/onju-voice-satellite/issues?q=is%3Aissue+label%3A%22bug%22+)..
- [ ] I am able to consult debug logging for my ESPHome instance.

### Describe the issue

I just flashed the latest version of the yaml but now I no longer set volume control response on the unit itslef.
The left/right touch buttons work and its indeed changing the volume but no visual feedback like before.
Whats weird is that when changing volume and then issue a voice command when I get the answer for a second the current volume is lighting up.

### Reproduction steps

1. Change volume via touch on device
2. No visual response of current volume level
3. Asking a question
4. When receiving answer current volume level lights up for 1-2 seconds

Mugga6315, 2024-05-02:
Okay, I debugged a little and compared old code with new one.
Problems seems to be the using `on_state` for the mediaplayer. Its said in the documentation:
_This trigger is activated each time the state of the media player is updated (for example, if the player is stop playing audio or received some command)._
This expains the weird behaviour why volume level is shown when speaker is talking.
In the old yaml the `show_volume` was triggered whitin the `set_volume` .

tetele, 2024-06-03:
After some investigation, it seems that the cause is [this ESPHome issue](https://github.com/esphome/issues/issues/5877).

If you try to set the volume when the media player is playing something longer, the visual representation works flawlessly.

tetele, 2024-06-04:
Can you try adding this to the config and checking if it solves the issue?

```yaml
external_components:
  - source: github://pr#6859
    components: i2s_audio
    refresh: 0s
```

Mugga6315, 2024-06-04:
Thx, for the reply. Visual feedback for volume works with that.
But sadly after any of the last esphome updates, the satellite isn't detecting any wakewords after the first command.

tetele, 2024-06-04:
Right, that's because of ESPHome 2024.5. The fix is [here](https://github.com/esphome/esphome/pull/6849) (applicable [like this](https://github.com/esphome/issues/issues/5794#issuecomment-2143787779)), but you can't have both that fix and this one simultaneously. It's a waiting game, sadly.

Mugga6315, 2024-06-04:
Okay, got it. But thats fine for me. I can just revert to the old volume setting stuff I mentioned and use the fix for working wakewords, for the moment.
Thanks

tetele, 2024-06-12:
Thanks for confirming. This was the fix, indeed https://github.com/esphome/esphome/pull/6859

tetele, 2024-06-21:
Fixed in ESPHome 2024.6 https://github.com/esphome/esphome/pull/6859

closed 2024-06-21 by tetele