tetele/onju-voice-satellite #46: media_player functionality using this config stopped working as of esphome 2024.5

rccoleman, 2024-05-15:
### Flavor

MicroWakeWord

### Checklist

- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/tetele/onju-voice-satellite/issues?q=is%3Aissue+label%3A%22bug%22+)..
- [X] I am able to consult debug logging for my ESPHome instance.

### Describe the issue

I've been happily using the media_player support introduced with https://github.com/tetele/onju-voice-satellite/pull/39 for a while now to replace my Alexa devices for TTS and it no longer works with the just-released ESPHome 2024.5.  I noticed this during the 2024.5 beta, but didn't have a chance to fully investigate.  Media player supports starts working again when I go back to 2024.4.

### Reproduction steps

Flash an Onju voice device with the config from [here](https://github.com/tetele/onju-voice-satellite/blob/main/esphome/onju-voice-microwakeword.yaml) and this diff to pull in the latest microwakeword changes:

```
diff --git a/esphome/onju-voice-microwakeword.yaml b/esphome/onju-voice-microwakeword.yaml
index a2322fe..c530026 100644
--- a/esphome/onju-voice-microwakeword.yaml
+++ b/esphome/onju-voice-microwakeword.yaml
@@ -9,6 +9,10 @@ external_components:
       url: https://github.com/gnumpi/esphome_audio
       ref: main
     components: [ adf_pipeline, i2s_audio ]
+    refresh: 0s
+  - source: github://pr#6653
+    components: [micro_wake_word]
+    refresh: 0s
 
 esphome:
   name: "${name}"
@@ -196,9 +200,6 @@ media_player:
             old_volume = new_volume;
 
 micro_wake_word:
-  model: okay_nabu
-  # model: hey_jarvis
-  # model: alexa
   on_wake_word_detected:
     - if:
         condition: media_player.is_playing
@@ -206,6 +207,14 @@ micro_wake_word:
           - media_player.pause
     - voice_assistant.start:
         wake_word: !lambda return wake_word;
+  vad_model:
+    model: https://github.com/kahrendt/microWakeWord/releases/download/model/vad_model.json
+    sliding_window_average_size: 2
+    threshold:
+      upper: 0.95
+      lower: 0.5
+  models:
+    - model: okay_nabu
 
 voice_assistant:
   id: va
```

### Debug logs

```text
[10:00:00][I][app:100]: ESPHome version 2024.5.0 compiled on May 15 2024, 09:54:07
[10:00:00][I][app:102]: Project tetele.onju_voice_satellite version 1.0.0
[10:00:00][C][wifi:580]: WiFi:
[10:00:00][C][wifi:408]:   Local MAC: 80:65:99:A2:AD:A4
[10:00:00][C][wifi:413]:   SSID: [redacted]
[10:00:00][C][wifi:416]:   IP Address: 192.168.1.131
[... 292 lines cut ...]

[10:05:47][D][esp-idf:000][http]: I (356102) HTTP_STREAM: total_bytes=0

[10:05:47][D][esp-idf:000][http]: E (356106) HTTP_STREAM: Invalid HTTP stream, status code = 500

[10:05:47][D][esp-idf:000][http]: E (356110) AUDIO_ELEMENT: [http] AEL_STATUS_ERROR_OPEN,-1

[10:05:47][D][esp-idf:000][http]: W (356114) AUDIO_ELEMENT: [http] audio_element_on_cmd_error,7

[10:05:47][D][esp-idf:000][decoder]: W (356122) AUDIO_ELEMENT: IN-[decoder] AEL_IO_ABORT

[10:05:47][D][esp-idf:000][decoder]: E (356127) MP3_DECODER: failed to read audio data (line 117)

[10:05:47][D][esp-idf:000][decoder]: W (356131) AUDIO_ELEMENT: [decoder] AEL_IO_ABORT, -3

[10:05:47][D][esp-idf:000][decoder]: I (356136) MP3_DECODER: Closed

[10:05:54][I][esp_adf_pipeline:185]: Pipeline preparation timeout!
[10:05:54][D][esp-idf:000]: W (363194) AUDIO_PIPELINE: Without stop, st:1

[10:05:54][D][esp_adf_pipeline:302]: State changed from PREPARING to STOPPING
[10:05:54][I][adf_media_player:135]: got new pipeline state: 4
[10:05:54][D][esp_adf_pipeline:302]: State changed from STOPPING to STOPPED
[10:05:54][I][adf_media_player:135]: got new pipeline state: 5
```

nyok92, 2024-05-16:
Same issue here

rccoleman, 2024-05-16:
I tried playing a simple WAV file and it also failed.  Looks like its trying to decode it as an MP3?

```
[08:43:48][D][media_player:061]: 'Onju Voice Satellite Dining Room' - Setting
[08:43:48][D][media_player:068]:   Media URL: https://xxx/media/local/broadcast.wav?authSig=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiIxNGUwYmQzMmVkZWM0ZjA4YWQwNjlkNjMyNzE4YWI5OSIsInBhdGgiOiIvbWVkaWEvbG9jYWwvYnJvYWRjYXN0LndhdiIsInBhcmFtcyI6W10sImlhdCI6MTcxNTg3NDIyOCwiZXhwIjoxNzE1OTYwNjI4fQ.fJvCswILFkyuPwHHoJU-shUs79ZXdUCV6gaVtNf4aTw
[08:43:48][D][adf_media_player:030]: Got control call in state 1
[08:43:48][D][esp_adf_pipeline:050]: Starting request, current state STOPPED
[08:43:48][D][esp_adf_pipeline:302]: State changed from STOPPED to PREPARING
[08:43:48][I][adf_media_player:135]: got new pipeline state: 1
[08:43:48][D][adf_i2s_out:127]: Set final i2s settings: 16000
[08:43:48][D][esp_audio_processors:079]: New settings: SRC: rate: 24000, ch: 1 DST: rate: 16000, ch: 2 
[08:43:48][D][esp-idf:000][http]: I (109097) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:1

[08:43:48][D][esp-idf:000][decoder]: I (109099) AUDIO_ELEMENT: [decoder] AEL_MSG_CMD_RESUME,state:1

[08:43:48][D][esp_audio_sources:097]: Streamer status: 2
[08:43:48][D][esp_audio_sources:098]: decoder status: 2
[08:43:51][D][esp-idf:000][http]: I (112010) HTTP_CLIENT: Body received in fetch header state, 0x3fcc91cc, 1708

[08:43:51][D][esp-idf:000][http]: I (112014) HTTP_STREAM: total_bytes=44966

[08:43:51][D][esp-idf:000][http]: W (112124) HTTP_STREAM: No more data,errno:0, total_bytes:44966, rlen = 0

[08:43:51][D][esp-idf:000][http]: I (112127) AUDIO_ELEMENT: IN-[http] AEL_IO_DONE,0

[08:43:51][D][esp-idf:000][decoder]: I (112132) AUDIO_ELEMENT: IN-[decoder] AEL_IO_DONE,-2

[08:43:51][D][esp-idf:000][decoder]: E (112136) MPEG_READER: resync error (line 405)

[08:43:51][D][esp-idf:000][decoder]: E (112140) MP3_DECODER: Encountered error reading when MP3 init

[08:43:51][D][esp-idf:000][decoder]: E (112144) AUDIO_ELEMENT: [decoder] AEL_STATUS_ERROR_OPEN,-1

[08:43:51][D][esp-idf:000][decoder]: W (112148) AUDIO_ELEMENT: [decoder] audio_element_on_cmd_error,7

[08:43:51][D][esp-idf:000][decoder]: I (112153) MP3_DECODER: Closed

[08:43:58][I][esp_adf_pipeline:185]: Pipeline preparation timeout!
[08:43:58][D][esp-idf:000]: W (119081) AUDIO_PIPELINE: Without stop, st:1

[08:43:58][D][esp_adf_pipeline:302]: State changed from PREPARING to STOPPING
[08:43:58][I][adf_media_player:135]: got new pipeline state: 4
[08:43:58][D][esp_adf_pipeline:302]: State changed from STOPPING to STOPPED
[08:43:58][I][adf_media_player:135]: got new pipeline state: 5
```

As before, the URL plays fine in a browser.  Probably related to https://github.com/esphome/issues/issues/5791.

tbrasser, 2024-05-17:
I changed from 32bit to 16bit output and changed the bitrate to 44100, matched that in music assistant player settings and now audio output is back!

- using non-mww
- using new esphome audio from gnumpi
- using dev-next branch
- latest MA / Esphome

Somehow the media_player doesn't know it's playing, but it is and also volume control works perfectly.

I'll try 24bits per sample and 48000 samplerate next

nyok92, 2024-05-17:
Hi @tbrasser ,
Could you share your ESP config ?
tks

tbrasser, 2024-05-17:
https://github.com/tbrasser/config/blob/main/esphome/includes/onju-voice.yaml

nyok92, 2024-05-17:
ok thanks @tbrasser 
using non-mww : what is it ?

tbrasser, 2024-05-17:
Openwakeword instead of microwakeword

tbrasser, 2024-05-17:
Basically I've grown to like using a custom wakeword but still have to train it longer, before thinking about trying mww. Also I have tablets running android ip camera apps rtsp streaming (via frigate) into streamAssist, so 1 or 2 extra openwakeword streams behind a local vad (or triggered via frigate motion/sound detection) doesn't hurt much.

rccoleman, 2024-05-18:
This fixes it for me: https://github.com/esphome/issues/issues/5791#issuecomment-2118573396.  Looks like bumping `esp-idf` to `4.47` is problematic.

tbrasser, 2024-05-20:
So I cannot get sound output on .7 only on recommended version. Wthell...

Probably because of the esphome_audio branch I'm using @tetele any ideas?

Also, what samplerates & bit per sample does this support? I understand it (tries to) resample when there's a mismatch, but since I'm using music assistant I can match it on that end I'd like to configure everything nicely.

Until now I only got (correct) output when setting 44100 Hz & 16bit.

rccoleman, 2024-05-20:
`4.4.7` is the new recommended version as of ESPHome 2024.5, and I needed to downgrade to `4.4.6` to play audio through the `media_player` entity.  I've been able to play 24KHz/32-bit TTS and 44.1KHz/32-bit MP3 content with just this change:

```
esp32:
  board: esp32-s3-devkitc-1
   framework:
     type: esp-idf
-    version: recommended
+    version: 4.4.6
```

gnumpi, 2024-05-23:
Some background info:
The cause of all these problems is the messy i2s driver of IDF-4.* which has no way of setting different channel formats for RX and TX. This will change in IDF-5 (while probably introducing a bunch of new problems). 
That's why the duplex mode of https://github.com/gnumpi/esphome_audio was meant to be configured such, that both, TX and RX are set to the same maximum needed quality (e.g. stereo, 48kHz, 16bit) and let the re-samplers do the work of converting the actual audio to the needed format. 
But, as discussed here https://github.com/gnumpi/esphome_audio/issues/17, mWW seems to rely on the volume increase introduced by the microphone component when converting 32bit audio to the desired 16bit. This volume increase doesn't happen when the re-sampler is converting it beforehand. A dirty hack was to simply set both TX and RX i2s settings to the ones needed by the microphone and remove the re-sampler from the microphone pipeline.
With IDF-4.4.7 the impact of setting the channel format on the TX direction changed, which causes the hack not to work anymore. 
What you can try is setting the channel config of audio_in to *all_right* instead of *right*. But **if** it's working, its just another hack. 
Unfortunately, I haven't fully understand the impact of the change in 4.4.7 yet and there is no documentation at all. To get a better understanding what's going on, @rccoleman could you share your config and logs when playing 24kHz/32-bit mono TTS with 4.4.6 and 4.4.7? Thanks!

s00500, 2024-05-23:
Just ran into this very same issue, trying the downgrade to .6 today, might later be able to do some testing of the channel config, thanks for all your awesome work here!

rccoleman, 2024-05-23:
> To get a better understanding what's going on, @rccoleman could you share your config and logs when playing 24kHz/32-bit mono TTS with 4.4.6 and 4.4.7? Thanks!

Sure, will try to collect those later today.

gnumpi, 2024-05-24:
Thanks a lot for doing the tests and sharing the logs. I think I do understand now what's going on here. 
The onju only has one speaker, right? So setting the output also to mono will be sufficient, right?
Can anyone with an onju try to test setting the channel of the adf-output also to *right*:
```yaml
adf_pipeline:
  - platform: i2s_audio
    type: audio_out
    id: adf_i2s_out
    i2s_audio_id: i2s_shared
    i2s_dout_pin: GPIO12
    sample_rate: 16000
    adf_alc: true
    bits_per_sample: 32bit
    fixed_settings: true
    channel: right

  - platform: i2s_audio
    type: audio_in
    id: adf_i2s_in
    i2s_audio_id: i2s_shared
    i2s_din_pin: GPIO17
    channel: right
    pdm: false
    sample_rate: 16000
    bits_per_sample: 32bit
    fixed_settings: true
``` 

Unfortunately, there was also a bug in the ADF-I2SReader, you might also want to try using the dev version of the ADF-Pipeline:
```yaml
external_components:
  - source:
      type: git
      url: https://github.com/gnumpi/esphome_audio
      ref: dev-next
```

nyok92, 2024-05-24:
With the previous config (esp-idf version 4.4.7) with dev-next & Chanel: right changes : STT and intents are working on my setup but audio out (TTS or music from music from Music assistant is not working.

Here are the logs when trying to play music from MA (enforce MP3 is set) :

```
INFO ESPHome 2024.5.2
INFO Reading configuration /config/esphome/nest-mini.yaml...
INFO Starting log output from XXX using esphome API
INFO Successfully connected to nest-mini @ XXX in 0.142s
INFO Successful handshake with nest-mini @ XXX in 0.068s
[14:25:18][I][app:100]: ESPHome version 2024.5.2 compiled on May 24 2024, 14:14:49
[14:25:18][I][app:102]: Project tetele.onju_voice_satellite version 1.0.0
[14:25:18][C][wifi:580]: WiFi:
[14:25:18][C][wifi:408]:   Local MAC: XXX
[14:25:18][C][wifi:413]:   SSID: [redacted]
[14:25:18][C][wifi:416]:   IP Address: XXX
[14:25:18][C][wifi:420]:   BSSID: [redacted]
[14:25:18][C][wifi:421]:   Hostname: 'nest-mini'
[14:25:18][C][wifi:423]:   Signal strength: -59 dB ▂▄▆█
[14:25:18][V][wifi:425]:   Priority: 0.0
[14:25:18][C][wifi:427]:   Channel: 6
[14:25:18][C][wifi:428]:   Subnet: 255.255.255.0
[14:25:18][C][wifi:429]:   Gateway: 192.168.1.254
[14:25:18][C][wifi:430]:   DNS1: 1.1.1.1
[14:25:18][C][wifi:431]:   DNS2: 1.0.0.1
[14:25:18][C][logger:185]: Logger:
[14:25:18][C][logger:186]:   Level: VERBOSE
[14:25:18][C][logger:188]:   Log Baud Rate: 115200
[14:25:18][C][logger:189]:   Hardware UART: USB_SERIAL_JTAG
[14:25:18][C][logger:193]:   Level for 'micro_wake_word': DEBUG
[14:25:18][C][template.number:050]: Template Number 'Touch threshold percentage'
[14:25:18][C][template.number:051]:   Optimistic: YES
[14:25:18][C][template.number:052]:   Update Interval: never
[14:25:18][C][esp32_rmt_led_strip:175]: ESP32 RMT LED Strip:
[14:25:18][C][esp32_rmt_led_strip:176]:   Pin: 11
[14:25:18][C][esp32_rmt_led_strip:177]:   Channel: 0
[14:25:18][C][esp32_rmt_led_strip:202]:   RGB Order: GRB
[14:25:18][C][esp32_rmt_led_strip:203]:   Max refresh rate: 0
[14:25:18][C][esp32_rmt_led_strip:204]:   Number of LEDs: 6
[14:25:18][C][switch.gpio:068]: GPIO Switch 'dac_mute'
[14:25:18][C][switch.gpio:091]:   Restore Mode: always OFF
[14:25:18][C][switch.gpio:031]:   Pin: GPIO21
[14:25:18][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Disable wake word'
[14:25:18][C][gpio.binary_sensor:016]:   Pin: GPIO38
[14:25:18][C][light:103]: Light 'leds'
[14:25:18][C][light:105]:   Default Transition Length: 0.0s
[14:25:18][C][light:106]:   Gamma Correct: 2.80
[14:25:18][C][light:103]: Light 'left_led'
[14:25:18][C][light:105]:   Default Transition Length: 0.1s
[14:25:18][C][light:106]:   Gamma Correct: 2.80
[14:25:18][C][light:103]: Light 'top_led'
[14:25:18][C][light:105]:   Default Transition Length: 0.1s
[14:25:18][C][light:106]:   Gamma Correct: 2.80
[14:25:18][C][light:103]: Light 'right_led'
[14:25:18][C][light:105]:   Default Transition Length: 0.1s
[14:25:18][C][light:106]:   Gamma Correct: 2.80
[14:25:18][C][template.switch:068]: Template Switch 'Use Wake Word'
[14:25:18][C][template.switch:091]:   Restore Mode: restore defaults to ON
[14:25:18][C][template.switch:057]:   Optimistic: YES
[14:25:18][C][psram:020]: PSRAM:
[14:25:18][C][psram:021]:   Available: YES
[14:25:18][C][psram:024]:   Size: 8191 KB
[14:25:18][C][i2s_audio:028]: I2SController:
[14:25:18][C][i2s_audio:029]:   AccessMode: duplex
[14:25:18][C][i2s_audio:030]:   Port: 0
[14:25:18][C][i2s_audio:032]:   Reader registered.
[14:25:18][C][i2s_audio:035]:   Writer registered.
[14:25:18][C][i2s_audio:139]: I2S-Writer (Fixed-CFG):
[14:25:18][C][i2s_audio:141]:   sample-rate: 16000 bits_per_sample: 32
[14:25:18][C][i2s_audio:142]:   channel_fmt: 3 channels: 1
[... 428 lines cut ...]
[16:31:04][D][esp_adf_pipeline:437]: [ADFMicrophone] Pipeline changed from ABORTING to STOPPED. (REQ: 1)
[16:31:04][D][voice_assistant:439]: State changed from STOPPING_MICROPHONE to AWAITING_RESPONSE
[16:31:04][D][voice_assistant:563]: Event Type: 0
[16:31:04][E][voice_assistant:693]: Error: stt-no-text-recognized - No text recognized
[16:31:04][D][voice_assistant:556]: Signaling stop...
[16:31:04][D][voice_assistant:439]: State changed from AWAITING_RESPONSE to STOP_MICROPHONE
[16:31:04][D][voice_assistant:445]: Desired state set to IDLE
[16:31:04][D][voice_assistant:439]: State changed from STOP_MICROPHONE to IDLE
[16:31:04][D][light:036]: 'top_led' Setting:
[16:31:04][D][light:059]:   Red: 100%, Green: 0%, Blue: 0%
[16:31:04][D][light:085]:   Transition length: 0.1s
[16:31:04][D][light:091]:   Effect: 'None'
[16:31:04][D][voice_assistant:563]: Event Type: 2
[16:31:04][D][voice_assistant:653]: Assist Pipeline ended
[16:31:04][D][light:036]: 'top_led' Setting:
[16:31:04][D][light:047]:   State: OFF
[16:31:04][D][light:085]:   Transition length: 0.1s
[16:31:05][D][light:036]: 'top_led' Setting:
[16:31:05][D][light:085]:   Transition length: 0.1s
```

Is buffer limited because of hardware specs  ? or could we increase its value ?

Hope it will help to debug this issue
by the way thanks for all this work :)

gnumpi, 2024-05-24:
Thanks for testing, it showed that I missed adapting the bit-depth of the re-sampler correctly! I just uploaded a fix.

Yes, you are right in duplex mode the output and input sample_rate, bit_depth and number of channels must be the same.

rccoleman, 2024-05-24:
> Thanks a lot for doing the tests and sharing the logs. I think I do understand now what's going on here. The onju only has one speaker, right? So setting the output also to mono will be sufficient, [...]

I made those changes to my Onju config along to moving back to the `recommended` version of `esp-idf` and now I get no audio at all.  The lighting pattern appears to indicate that it's not even trying to speak the response (it just goes right back to the "listening for the wakeword" pattern), but it looks like it is trying to speak from the logs.

[logs_onju-voice-dr_logs.txt](https://github.com/tetele/onju-voice-satellite/files/15436558/logs_onju-voice-dr_logs.txt)
[onju-voice-microwakeword.yaml.txt](https://github.com/tetele/onju-voice-satellite/files/15436588/onju-voice-microwakeword.yaml.txt)

nyok92, 2024-05-24:
@rccoleman You have to select channel: left for audio_out

@gnumpi I cleaned build files and installed again but output sound is still acceelerated with this following config:

[onju-voice.txt](https://github.com/tetele/onju-voice-satellite/files/15436716/onju-voice.txt)

```
INFO ESPHome 2024.5.2
INFO Reading configuration /config/esphome/nest-mini.yaml...
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-adf.git@v2.5
INFO Updating submodules (components/esp-adf-libs, components/esp-sr) for https://github.com/espressif/esp-adf.git@v2.5
INFO Updating https://github.com/espressif/esp-tflite-micro@None
INFO Compiling app...
Processing nest-mini (board: esp32-s3-devkitc-1; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/Improv @ 1.2.3
INFO Installing esphome/Improv @ 1.2.3
Unpacking  [####################################]  100%
Library Manager: Improv@1.2.3 has been installed!
INFO Improv@1.2.3 has been installed!
Library Manager: Installing esphome/noise-c @ 0.1.4
INFO Installing esphome/noise-c @ 0.1.4
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.4 has been installed!
INFO noise-c@0.1.4 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.1
INFO Installing esphome/libsodium @ 1.10018.1
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.1 has been installed!
INFO libsodium@1.10018.1 has been installed!
/data/cache/platformio/packages/framework-espidf@3.40407.0 Already Patched: yes
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - framework-espidf @ 3.40407.0 (4.4.7) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Generating assembly for certificate bundle...
Dependency Graph
|-- Improv @ 1.2.3
|-- noise-c @ 0.1.4
Generating assembly for .pioenvs/nest-mini/duer_profile.S
Compiling .pioenvs/nest-mini/src/esphome/components/adf_pipeline/adf_audio_element.o
Compiling .pioenvs/nest-mini/src/esphome/components/adf_pipeline/adf_audio_process.o
Compiling .pioenvs/nest-mini/src/esphome/components/adf_pipeline/adf_audio_sinks.o
Compiling .pioenvs/nest-mini/src/esphome/components/adf_pipeline/adf_audio_sources.o
Compiling .pioenvs/nest-mini/src/esphome/components/adf_pipeline/adf_pipeline.o
Compiling .pioenvs/nest-mini/src/esphome/components/adf_pipeline/adf_pipeline_controller.o
Compiling .pioenvs/nest-mini/src/esphome/components/adf_pipeline/media_player/adf_media_player.o
Compiling .pioenvs/nest-mini/src/esphome/components/adf_pipeline/microphone/esp_adf_microphone.o
Compiling .pioenvs/nest-mini/src/esphome/components/api/api_connection.o
Compiling .pioenvs/nest-mini/src/esphome/components/api/api_frame_helper.o
Compiling .pioenvs/nest-mini/src/esphome/components/api/api_pb2.o
Compiling .pioenvs/nest-mini/src/esphome/components/api/api_pb2_service.o
Compiling .pioenvs/nest-mini/src/esphome/components/api/api_server.o
Compiling .pioenvs/nest-mini/src/esphome/components/api/list_entities.o
Compiling .pioenvs/nest-mini/src/esphome/components/api/proto.o
Compiling .pioenvs/nest-mini/src/esphome/components/api/subscribe_state.o
Compiling .pioenvs/nest-mini/src/esphome/components/api/user_services.o
Compiling .pioenvs/nest-mini/src/esphome/components/binary_sensor/automation.o
Compiling .pioenvs/nest-mini/src/esphome/components/binary_sensor/binary_sensor.o
Compiling .pioenvs/nest-mini/src/esphome/components/binary_sensor/filter.o
Compiling .pioenvs/nest-mini/src/esphome/components/captive_portal/captive_portal.o
Compiling .pioenvs/nest-mini/src/esphome/components/dashboard_import/dashboard_import.o
Compiling .pioenvs/nest-mini/src/esphome/components/esp32/core.o
[... 2285 lines cut ...]
[17:35:44][I][esp_adf_pipeline:124]: [ decoder ] status: 12
[17:35:44][I][HTTPStreamReader:193]: [ * ] Receive music info from mp3 decoder, sample_rates=44100, bits=16, ch=2
[17:35:46][D][media_player:061]: 'Onju Voice Satellite' - Setting
[17:35:47][D][media_player:065]:   Command: STOP
[17:35:47][D][esp_adf_pipeline:070]: Called 'stop' while in RUNNING state.
[17:35:47][D][esp_adf_pipeline:437]: [MediaPlayer] Pipeline changed from RUNNING to ABORTING. (REQ: 1)
[17:35:47][I][adf_media_player:189]: got new pipeline state: 10, while in MP state PLAYING
[17:35:47][I][adf_media_player:252]: current mp state: PLAYING
[17:35:47][I][adf_media_player:253]: anouncement: false
[17:35:47][I][adf_media_player:254]: play_intent: false
[17:35:47][I][adf_media_player:255]: current_uri_: false
[17:35:47][D][adf_audio_element:324]: [http] Checking State for stopping, got 3
[17:35:47][D][adf_audio_element:324]: [decoder] Checking State for stopping, got 3
[17:35:47][V][esp-idf:000][resampler]: W (35755) AUDIO_ELEMENT: IN-[resampler] AEL_IO_ABORT

[17:35:47][V][esp-idf:000][decoder]: I (35758) MP3_DECODER: Closed

[17:35:47][D][adf_audio_element:324]: [i2s_out] Checking State for stopping, got 3
[17:35:47][D][esp_adf_pipeline:437]: [MediaPlayer] Pipeline changed from ABORTING to STOPPED. (REQ: 1)
[17:35:47][I][adf_media_player:189]: got new pipeline state: 4, while in MP state PLAYING
[17:35:47][I][adf_media_player:252]: current mp state: IDLE
[17:35:47][I][adf_media_player:253]: anouncement: false
[17:35:47][I][adf_media_player:254]: play_intent: false
[17:35:47][I][adf_media_player:255]: current_uri_: false
```

gnumpi, 2024-05-24:
@nyok92 it seems it did not load the version with the latest fixes. Can you add a *refresh: 0s* to the external source?

nyok92, 2024-05-24:
ok it seeems to have beeen updated now.
But now with the newer version i don't have out audio anymore 

```
INFO ESPHome 2024.5.2
INFO Reading configuration /config/esphome/nest-mini.yaml...
INFO Updating https://github.com/gnumpi/esphome_audio@dev-next
INFO Starting log output from 192.168.1.132 using esphome API
INFO Successfully connected to nest-mini @ 192.168.1.132 in 0.215s
INFO Successful handshake with nest-mini @ 192.168.1.132 in 0.087s
[18:30:32][I][app:100]: ESPHome version 2024.5.2 compiled on May 24 2024, 18:18:50
[18:30:32][I][app:102]: Project tetele.onju_voice_satellite version 1.0.0
[18:30:32][C][wifi:580]: WiFi:
[18:30:32][C][wifi:408]:   Local MAC: XXX
[18:30:32][C][wifi:413]:   SSID: [redacted]
[18:30:32][C][wifi:416]:   IP Address: 192.168.1.132
[18:30:32][C][wifi:420]:   BSSID: [redacted]
[18:30:32][C][wifi:421]:   Hostname: 'nest-mini'
[18:30:32][C][wifi:423]:   Signal strength: -63 dB ▂▄▆█
[18:30:32][V][wifi:425]:   Priority: 0.0
[18:30:32][C][wifi:427]:   Channel: 6
[18:30:32][C][wifi:428]:   Subnet: 255.255.255.0
[18:30:32][C][wifi:429]:   Gateway: 192.168.1.254
[18:30:32][C][wifi:430]:   DNS1: 1.1.1.1
[18:30:32][C][wifi:431]:   DNS2: 1.0.0.1
[18:30:32][C][logger:185]: Logger:
[18:30:32][C][logger:186]:   Level: VERBOSE
[18:30:32][C][logger:188]:   Log Baud Rate: 115200
[18:30:32][C][logger:189]:   Hardware UART: USB_SERIAL_JTAG
[18:30:32][C][logger:193]:   Level for 'micro_wake_word': DEBUG
[18:30:32][C][template.number:050]: Template Number 'Touch threshold percentage'
[18:30:32][C][template.number:051]:   Optimistic: YES
[18:30:32][C][template.number:052]:   Update Interval: never
[18:30:32][C][esp32_rmt_led_strip:175]: ESP32 RMT LED Strip:
[18:30:32][C][esp32_rmt_led_strip:176]:   Pin: 11
[18:30:32][C][esp32_rmt_led_strip:177]:   Channel: 0
[18:30:32][C][esp32_rmt_led_strip:202]:   RGB Order: GRB
[18:30:32][C][esp32_rmt_led_strip:203]:   Max refresh rate: 0
[18:30:32][C][esp32_rmt_led_strip:204]:   Number of LEDs: 6
[18:30:32][C][switch.gpio:068]: GPIO Switch 'dac_mute'
[18:30:32][C][switch.gpio:091]:   Restore Mode: always OFF
[18:30:32][C][switch.gpio:031]:   Pin: GPIO21
[18:30:32][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Disable wake word'
[18:30:32][C][gpio.binary_sensor:016]:   Pin: GPIO38
[18:30:32][C][light:103]: Light 'leds'
[18:30:32][C][light:105]:   Default Transition Length: 0.0s
[18:30:32][C][light:106]:   Gamma Correct: 2.80
[18:30:32][C][light:103]: Light 'left_led'
[18:30:32][C][light:105]:   Default Transition Length: 0.1s
[18:30:32][C][light:106]:   Gamma Correct: 2.80
[18:30:32][C][light:103]: Light 'top_led'
[18:30:32][C][light:105]:   Default Transition Length: 0.1s
[18:30:32][C][light:106]:   Gamma Correct: 2.80
[18:30:32][C][light:103]: Light 'right_led'
[18:30:32][C][light:105]:   Default Transition Length: 0.1s
[18:30:32][C][light:106]:   Gamma Correct: 2.80
[18:30:32][C][template.switch:068]: Template Switch 'Use Wake Word'
[18:30:32][C][template.switch:091]:   Restore Mode: restore defaults to ON
[18:30:32][C][template.switch:057]:   Optimistic: YES
[18:30:32][C][psram:020]: PSRAM:
[18:30:32][C][psram:021]:   Available: YES
[18:30:32][C][psram:024]:   Size: 8191 KB
[18:30:32][C][i2s_audio:028]: I2SController:
[18:30:32][C][i2s_audio:029]:   AccessMode: duplex
[18:30:32][C][i2s_audio:030]:   Port: 0
[18:30:32][C][i2s_audio:032]:   Reader registered.
[18:30:32][C][i2s_audio:035]:   Writer registered.
[18:30:32][C][i2s_audio:139]: I2S-Writer (Fixed-CFG):
[18:30:32][C][i2s_audio:141]:   sample-rate: 16000 bits_per_sample: 32
[18:30:32][C][i2s_audio:142]:   channel_fmt: 4 channels: 1
[... 216 lines cut ...]
[18:30:46][V][esp-idf:000]: W (43664) AUDIO_ELEMENT: [decoder] Element has not create when AUDIO_ELEMENT_TERMINATE

[18:30:46][V][esp-idf:000]: W (43666) AUDIO_ELEMENT: [resampler] Element has not create when AUDIO_ELEMENT_TERMINATE

[18:30:46][V][esp-idf:000]: W (43668) AUDIO_ELEMENT: [i2s_out] Element has not create when AUDIO_ELEMENT_TERMINATE

[18:30:46][V][esp-idf:000]: I (43671) I2S: DMA queue destroyed

[18:30:46][V][esp-idf:000]: I (43673) I2S: DMA queue destroyed

[18:30:46][D][esp_adf_pipeline:437]: [MediaPlayer] Pipeline changed from DESTROYING to UNINITIALIZED. (REQ: 4)
[18:30:46][I][adf_media_player:189]: got new pipeline state: 0, while in MP state PLAYING
[18:30:46][I][adf_media_player:252]: current mp state: IDLE
[18:30:46][I][adf_media_player:253]: anouncement: false
[18:30:46][I][adf_media_player:254]: play_intent: false
[18:30:46][I][adf_media_player:255]: current_uri_: yes
[18:30:46][D][esp_adf_pipeline:437]: [MediaPlayer] Pipeline changed from UNINITIALIZED to UNINITIALIZED. (REQ: 4)
[18:30:46][I][adf_media_player:189]: got new pipeline state: 0, while in MP state IDLE
[18:30:46][I][adf_media_player:252]: current mp state: IDLE
[18:30:46][I][adf_media_player:253]: anouncement: false
[18:30:46][I][adf_media_player:254]: play_intent: false
[18:30:46][I][adf_media_player:255]: current_uri_: yes
[18:30:46][W][component:237]: Component adf_pipeline.media_player took a long time for an operation (1737 ms).
[18:30:46][W][component:238]: Components should block for at most 30 ms.
```

gnumpi, 2024-05-24:
I was just told that this config should be working with IDF 4.4.7 and the next-dev branch:

https://github.com/tbrasser/config/blob/main/esphome%2Fincludes%2Fonju-voice.yaml

tbrasser, 2024-05-25:
Kind of, mostly testing audio output, which seems to be working at 16bit 44100Hz from MA (also selected 16/44.1 there), the esphome media_player shows as playing and volume control works. On Music Assistant side it doesn't show as playing though.

Reading through above stuff seems I misconfigured my mic input so I'll try to test some more things this weekend.

gnumpi, 2024-06-07:
It seems to work now with the latest fixes in dev-next branch and this config:
```yaml
- source:
      type: git
      url: https://github.com/gnumpi/esphome_audio
      ref: dev-next
    components: [ adf_pipeline, i2s_audio ]
    refresh: 0s

adf_pipeline:
  - platform: i2s_audio
    type: audio_out
    id: adf_i2s_out
    i2s_audio_id: i2s_shared
    i2s_dout_pin: GPIO12
    sample_rate: 16000
    adf_alc: true
    bits_per_sample: 32bit
    fixed_settings: true
    channel: left
​
  - platform: i2s_audio
    type: audio_in
    id: adf_i2s_in
    i2s_audio_id: i2s_shared
    i2s_din_pin: GPIO17
    channel: left
    pdm: false
    sample_rate: 16000
    bits_per_sample: 32bit
    fixed_settings: true
​
​
microphone:
  - platform: adf_pipeline
    id: onju_microphone
    keep_pipeline_alive: true
    gain_log2: 3
    pipeline:
      - adf_i2s_in
      - self
​
media_player:
  - platform: adf_pipeline
    id: onju_out
    name: None
    internal: false
    keep_pipeline_alive: true
    pipeline:
      - self
      - resampler
      - adf_i2s_out

```

closed 2024-06-07 by tetele

nyok92, 2024-06-07:
Thanks, confirmed mmw/voice assist & media player are working correctly with duplex mode ! :) :) :)

PR #49 (2024-06-07): Fix audio on ESPHome 2024.5
Fix #46 with the updates made on the `dev-next` branch of [`esphome_audio`](https://github.com/gnumpi/esphome_audio) made by @gnumpi and the config changes proposed [here](https://github.com/tetele/onju-voice-satellite/issues/46#issuecomment-2154248036).

Thank you very much for these!