tetele/onju-voice-satellite #47: can not build onju-voice-microwakeword

lecrams, 2024-06-03:
### Flavor

MicroWakeWord

### Checklist

- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/tetele/onju-voice-satellite/issues?q=is%3Aissue+label%3A%22bug%22+)..
- [X] I am able to consult debug logging for my ESPHome instance.

### Describe the issue

When trying to build onju-voice-microwakeword I get the following error:

PACKAGES:
 - framework-espidf @ 3.40406.240122 (4.4.6)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40400.0 (4.4.0)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 2.35.0-20220830
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
ModuleNotFoundError: No module named 'pkg_resources':
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\platformio\builder\main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\a5128563\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 612:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\a5128563\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 279:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\a5128563\.platformio\platforms\espressif32\builder\main.py", line 312:
    target_elf = env.BuildProgram()
  File "C:\Users\a5128563\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Util\envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\platformio\builder\tools\piobuild.py", line 61:
    env.ProcessProgramDeps()
  File "C:\Users\a5128563\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Util\envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\platformio\builder\tools\piobuild.py", line 121:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\a5128563\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Util\envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\platformio\builder\tools\piobuild.py", line 347:
    SConscript(env.GetFrameworkScript(name), exports="env")
  File "C:\Users\a5128563\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 676:
    return method(*args, **kw)
  File "C:\Users\a5128563\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 612:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\a5128563\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 279:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\a5128563\.platformio\platforms\espressif32\builder\frameworks\espidf.py", line 29:
    import pkg_resources
============================================== [FAILED] Took 1.51 seconds ==============================================

### Reproduction steps

1. esphome compile onju-voice-microwakeword,.yaml
2.
3.
...

tetele, 2024-06-03:
Did you try to [clean the build files](https://esphome.io/guides/getting_started_hassio.html#dashboard-interface)?

lecrams, 2024-06-03:
I always start first time with a clean build. It gives exactly the same error message

tetele, 2024-06-04:
As far as I can work out, this is not an issue with the config, but with your esphome setup. I initially missed the fact that you've installed it in your Windows environment.

I'm not sure what exactly there is to do, but you can at least try to `pip install setuptools` and try again. Source: https://stackoverflow.com/a/10538412

closed 2024-06-04 by tetele

lecrams, 2024-06-04:
The installation of the setuptools solved the issue. It now builds okay!
Thx for the support