tetele/onju-voice-satellite #48: 2024.4.2 (out and some bugs) - Wake word from HA works, mic works, speaker does not

dreimer1986, 2024-06-04:
### Flavor

OpenWakeWord or no wake word

### Checklist

- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/tetele/onju-voice-satellite/issues?q=is%3Aissue+label%3A%22bug%22+)..
- [X] I am able to consult debug logging for my ESPHome instance.

### Describe the issue

New built Onju Voice. Tested a bit around and have Wake word working frine, according to logs, mic works fine, too and speaker does not... Well I tried around with media player before that and the speaker CAN output audio, so the hardware seems fine. Ah and when I used the wake word ONCE it does not work anymore afterwards

### Reproduction steps

1.
2.
3.
...

### Debug logs

```text
INFO ESPHome 2024.5.2
INFO Reading configuration /config/esphome/onju-voice-e27688.yaml...
INFO Starting log output from 192.168.181.103 using esphome API
INFO Successfully connected to onju-voice-e27688 @ 192.168.181.103 in 0.004s
INFO Successful handshake with onju-voice-e27688 @ 192.168.181.103 in 0.071s
[20:05:47][I][app:100]: ESPHome version 2024.5.2 compiled on Jun  4 2024, 19:26:14
[20:05:47][I][app:102]: Project tetele.onju_voice_satellite version 1.0.0
[20:05:47][C][wifi:580]: WiFi:
[20:05:47][C][wifi:408]:   Local MAC: 24:58:7C:E2:76:88
[20:05:47][C][wifi:413]:   SSID: 'Asgaard'[redacted]
[20:05:47][C][wifi:416]:   IP Address: 192.168.181.103
[20:05:47][C][wifi:420]:   BSSID: B0:F2:08:55:4B:1E[redacted]
[20:05:47][C][wifi:421]:   Hostname: 'onju-voice-e27688'
[20:05:47][C][wifi:423]:   Signal strength: -69 dB ▂▄▆█
[20:05:47][C][wifi:427]:   Channel: 13
[20:05:47][C][wifi:428]:   Subnet: 255.255.255.0
[20:05:47][C][wifi:429]:   Gateway: 192.168.181.254
[20:05:47][C][wifi:430]:   DNS1: 192.168.181.42
[20:05:47][C][wifi:431]:   DNS2: 0.0.0.0
[20:05:47][C][logger:185]: Logger:
[20:05:47][C][logger:186]:   Level: DEBUG
[20:05:47][C][logger:188]:   Log Baud Rate: 115200
[20:05:47][C][logger:189]:   Hardware UART: USB_CDC
[20:05:47][C][template.number:050]: Template Number 'Touch threshold percentage'
[20:05:47][C][template.number:051]:   Optimistic: YES
[20:05:47][C][template.number:052]:   Update Interval: never
[20:05:47][C][esp32_rmt_led_strip:175]: ESP32 RMT LED Strip:
[20:05:47][C][esp32_rmt_led_strip:176]:   Pin: 11
[20:05:47][C][esp32_rmt_led_strip:177]:   Channel: 0
[20:05:47][C][esp32_rmt_led_strip:202]:   RGB Order: GRB
[20:05:47][C][esp32_rmt_led_strip:203]:   Max refresh rate: 0
[20:05:47][C][esp32_rmt_led_strip:204]:   Number of LEDs: 6
[20:05:47][D][light:036]: 'top_led' Setting:
[20:05:47][D][light:051]:   Brightness: 60%
[20:05:47][D][light:059]:   Red: 100%, Green: 0%, Blue: 100%
[20:05:47][D][light:109]:   Effect: 'listening_ww'
[20:05:47][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Disable wake word'
[20:05:47][C][gpio.binary_sensor:016]:   Pin: GPIO38
[20:05:47][C][light:103]: Light 'leds'
[20:05:47][C][light:105]:   Default Transition Length: 0.0s
[20:05:47][C][light:106]:   Gamma Correct: 2.80
[20:05:47][C][light:103]: Light 'left_led'
[20:05:47][C][light:105]:   Default Transition Length: 0.1s
[20:05:47][C][light:106]:   Gamma Correct: 2.80
[20:05:47][C][light:103]: Light 'top_led'
[20:05:47][C][light:105]:   Default Transition Length: 0.1s
[... 71 lines cut ...]
[20:05:57][D][voice_assistant:563]: Event Type: 7
[20:05:57][D][voice_assistant:619]: Response: "Oh, ein Test? Lass uns sehen, ob ich deinen Anforderungen gerecht werden kann. 😉  Sprich, was möchtest du von mir wissen?"
[20:05:57][D][voice_assistant:563]: Event Type: 8
[20:05:57][D][voice_assistant:639]: Response URL: "https://192.168.181.42:8123/api/tts_proxy/3c20dbcdaf276fce00ce63a450b31ca4e60a9d58_de-de_70263bfdba_tts.home_assistant_cloud.mp3"
[20:05:57][D][voice_assistant:439]: State changed from AWAITING_RESPONSE to STREAMING_RESPONSE
[20:05:57][D][voice_assistant:445]: Desired state set to STREAMING_RESPONSE
[20:05:57][D][media_player:061]: 'Onju Voice Satellite e27688' - Setting
[20:05:57][D][media_player:068]:   Media URL: https://192.168.181.42:8123/api/tts_proxy/3c20dbcdaf276fce00ce63a450b31ca4e60a9d58_de-de_70263bfdba_tts.home_assistant_cloud.mp3
[20:05:57][D][media_player:074]:  Announcement: yes
[20:05:57][D][media_player:061]: 'Onju Voice Satellite e27688' - Setting
[20:05:57][D][media_player:068]:   Media URL: https://192.168.181.42:8123/api/tts_proxy/3c20dbcdaf276fce00ce63a450b31ca4e60a9d58_de-de_70263bfdba_tts.home_assistant_cloud.mp3
[20:05:57][D][light:036]: 'top_led' Setting:
[20:05:57][D][light:059]:   Red: 20%, Green: 100%, Blue: 0%
[20:05:57][D][light:109]:   Effect: 'speaking'
[20:05:57][D][voice_assistant:563]: Event Type: 2
[20:05:57][D][voice_assistant:653]: Assist Pipeline ended
[20:05:57][W][component:237]: Component i2s_audio.media_player took a long time for an operation (549 ms).
[20:05:58][W][component:238]: Components should block for at most 30 ms.
[20:05:58][W][component:237]: Component i2s_audio.media_player took a long time for an operation (477 ms).
[20:05:58][W][component:238]: Components should block for at most 30 ms.
[20:05:58][D][light:036]: 'top_led' Setting:
[20:05:58][D][light:051]:   Brightness: 60%
[20:05:58][D][light:059]:   Red: 100%, Green: 0%, Blue: 100%
[20:05:58][D][light:109]:   Effect: 'listening_ww'
```

dreimer1986, 2024-06-04:
Small addition: https://192.168.181.42:8123/api/tts_proxy/3c20dbcdaf276fce00ce63a450b31ca4e60a9d58_de-de_70263bfdba_tts.home_assistant_cloud.mp3 does not work in browser. but the same with http does.

Mugga6315, 2024-06-04:
Have exactly the same on my satellites after one of the last esphome updates.

tetele, 2024-06-04:
This is the issue https://github.com/esphome/issues/issues/5794

dreimer1986, 2024-06-04:
Yeah the wake word works only once problem indeed is this one, but the speaker still is stilent on 2024.4.2:

INFO ESPHome 2024.4.2
INFO Reading configuration /config/esphome/onju-voice-e27688.yaml...
INFO Cloning https://github.com/tetele/onju-voice-satellite.git@main
INFO Fetching main
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
Processing onju-voice-e27688 (board: esp32-s3-devkitc-1; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Platform Manager: Installing platformio/espressif32 @ 5.4.0
INFO Installing platformio/espressif32 @ 5.4.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: espressif32@5.4.0 has been installed!
INFO espressif32@5.4.0 has been installed!
Tool Manager: Installing espressif/toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
INFO Installing espressif/toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5 has been installed!
INFO toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5 has been installed!
Tool Manager: Installing espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
INFO Installing espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: toolchain-riscv32-esp@8.4.0+2021r2-patch5 has been installed!
INFO toolchain-riscv32-esp@8.4.0+2021r2-patch5 has been installed!
Tool Manager: Installing platformio/framework-arduinoespressif32 @ ~3.20005.0
INFO Installing platformio/framework-arduinoespressif32 @ ~3.20005.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: framework-arduinoespressif32@3.20005.220925 has been installed!
INFO framework-arduinoespressif32@3.20005.220925 has been installed!
Tool Manager: Installing platformio/tool-esptoolpy @ ~1.40400.0
INFO Installing platformio/tool-esptoolpy @ ~1.40400.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-esptoolpy@1.40400.0 has been installed!
INFO tool-esptoolpy@1.40400.0 has been installed!
Tool Manager: Installing platformio/tool-scons @ ~4.40600.0
INFO Installing platformio/tool-scons @ ~4.40600.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-scons@4.40600.0 has been installed!
INFO tool-scons@4.40600.0 has been installed!
Library Manager: Installing esphome/AsyncTCP-esphome @ 2.1.3
INFO Installing esphome/AsyncTCP-esphome @ 2.1.3
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@2.1.3 has been installed!
INFO AsyncTCP-esphome@2.1.3 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 3.1.0
INFO Installing esphome/ESPAsyncWebServer-esphome @ 3.1.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@3.1.0 has been installed!
INFO ESPAsyncWebServer-esphome@3.1.0 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/Improv @ 1.2.3
INFO Installing esphome/Improv @ 1.2.3
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: Improv@1.2.3 has been installed!
INFO Improv@1.2.3 has been installed!
Library Manager: Installing esphome/noise-c @ 0.1.4
INFO Installing esphome/noise-c @ 0.1.4
Downloading  [####################################]  100%
[... 587 lines cut ...]
[01:12:22][W][component:237]: Component i2s_audio.media_player took a long time for an operation (547 ms).
[01:12:22][W][component:238]: Components should block for at most 30 ms.
[01:12:23][W][component:237]: Component i2s_audio.media_player took a long time for an operation (479 ms).
[01:12:23][W][component:238]: Components should block for at most 30 ms.
[01:12:23][D][light:036]: 'top_led' Setting:
[01:12:23][D][light:051]:   Brightness: 60%
[01:12:23][D][light:059]:   Red: 100%, Green: 0%, Blue: 100%
[01:12:23][D][light:109]:   Effect: 'listening_ww'
[01:12:24][D][voice_assistant:439]: State changed from STREAMING_RESPONSE to IDLE
[01:12:24][D][voice_assistant:445]: Desired state set to IDLE
[01:12:24][D][voice_assistant:439]: State changed from IDLE to START_PIPELINE
[01:12:24][D][voice_assistant:445]: Desired state set to START_MICROPHONE
[01:12:24][D][voice_assistant:126]: microphone not running
[01:12:24][D][voice_assistant:210]: Requesting start...
[01:12:24][D][voice_assistant:439]: State changed from START_PIPELINE to STARTING_PIPELINE
[01:12:24][D][voice_assistant:126]: microphone not running
[01:12:24][D][voice_assistant:476]: Client started, streaming microphone
[01:12:24][D][voice_assistant:439]: State changed from STARTING_PIPELINE to START_MICROPHONE
[01:12:24][D][voice_assistant:445]: Desired state set to STREAMING_MICROPHONE
[01:12:24][D][voice_assistant:163]: Starting Microphone
[01:12:24][D][voice_assistant:439]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[01:12:24][D][voice_assistant:563]: Event Type: 1
[01:12:24][D][voice_assistant:566]: Assist Pipeline running
[01:12:24][D][voice_assistant:439]: State changed from STARTING_MICROPHONE to STREAMING_MICROPHONE
[01:12:24][D][voice_assistant:563]: Event Type: 9

closed 2024-06-05 by dreimer1986