tetele/onju-voice-satellite #51: Wake word detected only once.

nagydavid, 2024-06-09:
### Flavor

MicroWakeWord

### Checklist

- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/tetele/onju-voice-satellite/issues?q=is%3Aissue+label%3A%22bug%22+)..
- [X] I am able to consult debug logging for my ESPHome instance.

### Describe the issue

After installing the latest updated, wake word detection is happening only once, after boot up and after is dead.

### Reproduction steps

install microwakeword flavour.

### Debug logs

```text
INFO ESPHome 2024.5.5
INFO Reading configuration /config/esphome/voice-assist-living-room.yaml...
INFO Starting log output from 192.168.50.36 using esphome API
INFO Successfully connected to voice-assist-living-room @ 192.168.50.36 in 0.056s
INFO Successful handshake with voice-assist-living-room @ 192.168.50.36 in 0.056s
[21:50:43][I][app:100]: ESPHome version 2024.5.5 compiled on Jun  9 2024, 20:24:15
[21:50:43][I][app:102]: Project tetele.onju_voice_satellite version 1.0.0
[21:50:43][C][wifi:580]: WiFi:
[21:50:43][C][wifi:408]:   Local MAC: 68:B6:B3:20:C0:74
[21:50:43][C][wifi:413]:   SSID: [redacted]
[21:50:43][C][wifi:416]:   IP Address: 192.168.50.36
[21:50:43][C][wifi:420]:   BSSID: [redacted]
[21:50:43][C][wifi:421]:   Hostname: 'voice-assist-living-room'
[21:50:43][C][wifi:423]:   Signal strength: -53 dB ▂▄▆█
[21:50:43][C][wifi:427]:   Channel: 8
[21:50:43][C][wifi:428]:   Subnet: 255.255.255.0
[21:50:43][C][wifi:429]:   Gateway: 192.168.50.1
[21:50:43][C][wifi:430]:   DNS1: 0.0.0.0
[21:50:43][C][wifi:431]:   DNS2: 0.0.0.0
[21:50:43][C][logger:185]: Logger:
[21:50:43][C][logger:186]:   Level: DEBUG
[21:50:43][C][logger:188]:   Log Baud Rate: 115200
[21:50:43][C][logger:189]:   Hardware UART: USB_CDC
[21:50:43][C][template.number:050]: Template Number 'Touch threshold percentage'
[21:50:43][C][template.number:051]:   Optimistic: YES
[21:50:43][C][template.number:052]:   Update Interval: never
[21:50:43][C][esp32_rmt_led_strip:175]: ESP32 RMT LED Strip:
[21:50:43][C][esp32_rmt_led_strip:176]:   Pin: 11
[21:50:43][C][esp32_rmt_led_strip:177]:   Channel: 0
[21:50:43][C][esp32_rmt_led_strip:202]:   RGB Order: GRB
[21:50:43][C][esp32_rmt_led_strip:203]:   Max refresh rate: 0
[21:50:43][C][esp32_rmt_led_strip:204]:   Number of LEDs: 6
[21:50:43][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Disable wake word'
[21:50:43][C][gpio.binary_sensor:016]:   Pin: GPIO38
[21:50:43][C][light:103]: Light 'leds'
[21:50:43][C][light:105]:   Default Transition Length: 0.0s
[21:50:43][C][light:106]:   Gamma Correct: 2.80
[21:50:43][C][light:103]: Light 'left_led'
[21:50:43][C][light:105]:   Default Transition Length: 0.1s
[21:50:43][C][light:106]:   Gamma Correct: 2.80
[21:50:43][C][light:103]: Light 'top_led'
[21:50:43][C][light:105]:   Default Transition Length: 0.1s
[21:50:43][C][light:106]:   Gamma Correct: 2.80
[21:50:43][C][light:103]: Light 'right_led'
[21:50:43][C][light:105]:   Default Transition Length: 0.1s
[21:50:43][C][light:106]:   Gamma Correct: 2.80
[21:50:43][C][template.switch:068]: Template Switch 'Use Wake Word'
[21:50:43][C][template.switch:091]:   Restore Mode: restore defaults to ON
[21:50:43][C][template.switch:057]:   Optimistic: YES
[... 110 lines cut ...]
[21:51:03][D][voice_assistant:701]: Response URL: "http://192.168.50.188:8123/api/tts_proxy/85d43b448ab715eae17c0361864a34ff749eb14a_en-us_4e62813ccd_tts.piper.mp3"
[21:51:03][D][voice_assistant:502]: State changed from AWAITING_RESPONSE to STREAMING_RESPONSE
[21:51:03][D][voice_assistant:508]: Desired state set to STREAMING_RESPONSE
[21:51:03][D][media_player:061]: 'Voice Assist Living Room' - Setting
[21:51:03][D][media_player:068]:   Media URL: http://192.168.50.188:8123/api/tts_proxy/85d43b448ab715eae17c0361864a34ff749eb14a_en-us_4e62813ccd_tts.piper.mp3
[21:51:03][D][media_player:074]:  Announcement: yes
[21:51:03][D][media_player:061]: 'Voice Assist Living Room' - Setting
[21:51:03][D][media_player:068]:   Media URL: http://192.168.50.188:8123/api/tts_proxy/85d43b448ab715eae17c0361864a34ff749eb14a_en-us_4e62813ccd_tts.piper.mp3
[21:51:03][D][light:036]: 'top_led' Setting:
[21:51:03][D][light:059]:   Red: 20%, Green: 100%, Blue: 0%
[21:51:03][D][light:109]:   Effect: 'speaking'
[21:51:03][D][voice_assistant:625]: Event Type: 2
[21:51:03][D][voice_assistant:715]: Assist Pipeline ended
[21:51:04][W][component:237]: Component i2s_audio.media_player took a long time for an operation (524 ms).
[21:51:04][W][component:238]: Components should block for at most 30 ms.
[21:51:06][W][component:237]: Component i2s_audio.media_player took a long time for an operation (371 ms).
[21:51:06][W][component:238]: Components should block for at most 30 ms.
[21:51:06][W][component:237]: Component i2s_audio.media_player took a long time for an operation (369 ms).
[21:51:06][W][component:238]: Components should block for at most 30 ms.
[21:51:06][D][light:036]: 'top_led' Setting:
[21:51:06][D][light:051]:   Brightness: 60%
[21:51:06][D][light:059]:   Red: 100%, Green: 0%, Blue: 100%
[21:51:06][D][light:109]:   Effect: 'listening_ww'
```
```

tetele, 2024-06-10:
Seems like a duplicate of #46. Please reopen if that's not the case.

nagydavid, 2024-06-10:
So  I think it is not the same problem.  After the first wakeword detected and command executed, after the "warning" nothing works. Same applie, if wake word is disabled and I trigger the pipline by touch.

tetele, 2024-06-10:
Can you please try [this fix](https://github.com/tetele/onju-voice-satellite/issues/46#issuecomment-2154248036) and see if it works?

nagydavid, 2024-06-10:
I have tried it, the also with cleaned build files. I reinstalled openwakeword, and the same happens. After the first wake word detection, I get this warning 

```
[13:06:05][W][component:237]: Component i2s_audio.media_player took a long time for an operation (525 ms).
[13:06:05][W][component:238]: Components should block for at most 30 ms.
[13:06:07][W][component:237]: Component i2s_audio.media_player took a long time for an operation (371 ms).
[13:06:07][W][component:238]: Components should block for at most 30 ms.
[13:06:08][W][component:237]: Component i2s_audio.media_player took a long time for an operation (370 ms).
[13:06:08][W][component:238]: Components should block for at most 30 ms.
[13:06:08][D][light:036]: 'top_led' Setting:
[13:06:08][D][light:051]:   Brightness: 60%
[13:06:08][D][light:059]:   Red: 100%, Green: 0%, Blue: 100%
[13:06:08][D][light:109]:   Effect: 'listening_ww'
```
and nothing happens.

nagydavid, 2024-06-10:
I have reverted back to esphome 2023.4.2. Openwakeword pipeline works, microwakeword does not

VivantSenior, 2024-07-11:
I noticed that my instance of Onju with MWW reacts for wake word only two times. After that in need to uncheck and check "Use wake word" in order to be able to use wake word again.  ESPHome 2024.6.6.