tetele/onju-voice-satellite #53: version remains at ESPHome 2024.2.0 after recompiling and updating

rosscullen, 2024-06-14:
### Flavor

OpenWakeWord or no wake word

### Checklist

- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/tetele/onju-voice-satellite/issues?q=is%3Aissue+label%3A%22bug%22+)..
- [ ] I am able to consult debug logging for my ESPHome instance.

### Describe the issue

Hi there, using latest version of HA and ESPHome.

After recompiling and updaeding again, version remains at ESPHome 2024.2.0 and prompting me to update (even though i did). Using config on onju-voice-microwakeword.yaml (except for different hostname and WiFi settings). Any suggestiions of what silly step i'm missing would be greatly appreciated. Thanks

### Reproduction steps

1. Tried updating using the update option in ESPHome
2. Tried manual download and uploading
3. rebooted everything
...

### Debug logs

```text
Logs onju-voice-1.yaml
INFO ESPHome 2024.5.5
INFO Reading configuration /config/esphome/onju-voice-1.yaml...
INFO Starting log output from 192.168.9.15 using esphome API
INFO Successfully connected to onju-voice-1 @ 192.168.9.15 in 0.007s
INFO Successful handshake with onju-voice-1 @ 192.168.9.15 in 0.059s
[00:20:56][I][app:102]: ESPHome version 2024.2.0 compiled on Feb 25 2024, 12:33:07
[00:20:56][C][wifi:577]: WiFi:
```

tetele, 2024-06-17:
That means that the compile/upload has failed. Post-boot logs don't say anything about the cause.

Go to the Onju device and select Install > Wirelessly and look at the compilation log. The issue should be listed there. If the compilation fails, in the ESPHome dashboard click the 3 dot menu on the Onju device and select "Clean build files", then try another reinstall.

Again, without an error message, there's no way to tell what went wrong.

rosscullen, 2024-06-17:
Thanks for reply @tetele. I did a "Clean build files" and recompiled... but no success unfortunetely.

Below is a copy of the full logs. Would appreciate if you or group could review and suggest potential fixes (thanks):

`INFO ESPHome 2024.5.5
INFO Reading configuration /config/esphome/onju-voice-1.yaml...
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-adf.git@v2.5
INFO Updating submodules (components/esp-adf-libs, components/esp-sr) for https://github.com/espressif/esp-adf.git@v2.5
INFO Updating https://github.com/espressif/esp-tflite-micro@None
INFO Compiling app...
Processing onju-voice-1 (board: esp32-s3-devkitc-1; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/Improv @ 1.2.3
INFO Installing esphome/Improv @ 1.2.3
Unpacking  [####################################]  100%
Library Manager: Improv@1.2.3 has been installed!
INFO Improv@1.2.3 has been installed!
Library Manager: Installing esphome/noise-c @ 0.1.4
INFO Installing esphome/noise-c @ 0.1.4
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.4 has been installed!
INFO noise-c@0.1.4 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.1
INFO Installing esphome/libsodium @ 1.10018.1
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.1 has been installed!
INFO libsodium@1.10018.1 has been installed!
/data/cache/platformio/packages/framework-espidf Already Patched: yes
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - framework-espidf @ 3.40407.240606 (4.4.7) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Generating assembly for certificate bundle...
Dependency Graph
|-- Improv @ 1.2.3
|-- noise-c @ 0.1.4
Generating assembly for .pioenvs/onju-voice-1/duer_profile.S
Compiling .pioenvs/onju-voice-1/src/esphome/components/adf_pipeline/adf_audio_element.o
Compiling .pioenvs/onju-voice-1/src/esphome/components/adf_pipeline/adf_audio_process.o
Compiling .pioenvs/onju-voice-1/src/esphome/components/adf_pipeline/adf_audio_sinks.o
Compiling .pioenvs/onju-voice-1/src/esphome/components/adf_pipeline/adf_audio_sources.o
Compiling .pioenvs/onju-voice-1/src/esphome/components/adf_pipeline/adf_pipeline.o
Compiling .pioenvs/onju-voice-1/src/esphome/components/adf_pipeline/adf_pipeline_controller.o
Compiling .pioenvs/onju-voice-1/src/esphome/components/adf_pipeline/media_player/adf_media_player.o
Compiling .pioenvs/onju-voice-1/src/esphome/components/adf_pipeline/microphone/esp_adf_microphone.o
Compiling .pioenvs/onju-voice-1/src/esphome/components/api/api_connection.o
Compiling .pioenvs/onju-voice-1/src/esphome/components/api/api_frame_helper.o
Compiling .pioenvs/onju-voice-1/src/esphome/components/api/api_pb2.o
Compiling .pioenvs/onju-voice-1/src/esphome/components/api/api_pb2_service.o
Compiling .pioenvs/onju-voice-1/src/esphome/components/api/api_server.o
Compiling .pioenvs/onju-voice-1/src/esphome/components/api/list_entities.o
Compiling .pioenvs/onju-voice-1/src/esphome/components/api/proto.o
Compiling .pioenvs/onju-voice-1/src/esphome/components/api/subscribe_state.o
Compiling .pioenvs/onju-voice-1/src/esphome/components/api/user_services.o
Compiling .pioenvs/onju-voice-1/src/esphome/components/binary_sensor/automation.o
Compiling .pioenvs/onju-voice-1/src/esphome/components/binary_sensor/binary_sensor.o
Compiling .pioenvs/onju-voice-1/src/esphome/components/binary_sensor/filter.o
Compiling .pioenvs/onju-voice-1/src/esphome/components/captive_portal/captive_portal.o
Compiling .pioenvs/onju-voice-1/src/esphome/components/dashboard_import/dashboard_import.o
Compiling .pioenvs/onju-voice-1/src/esphome/components/esp32/core.o
Compiling .pioenvs/onju-voice-1/src/esphome/components/esp32/gpio.o
Compiling .pioenvs/onju-voice-1/src/esphome/components/esp32/preferences.o
Compiling .pioenvs/onju-voice-1/src/esphome/components/esp32_rmt_led_strip/led_strip.o
[... 2167 lines cut ...]
[23:13:47][D][voice_assistant:524]: Assist Pipeline running
[23:13:47][D][voice_assistant:521]: Event Type: 9
[23:13:47][D][light:036]: 'top_led' Setting:
[23:13:47][D][light:051]:   Brightness: 60%
[23:13:47][D][light:059]:   Red: 100%, Green: 0%, Blue: 100%
[23:13:47][D][light:085]:   Transition length: 0.1s
[23:13:52][D][voice_assistant:521]: Event Type: 0
[23:13:52][D][voice_assistant:521]: Event Type: 2
[23:13:52][D][voice_assistant:611]: Assist Pipeline ended
[23:13:52][D][voice_assistant:414]: State changed from STREAMING_MICROPHONE to IDLE
[23:13:52][D][voice_assistant:420]: Desired state set to IDLE
[23:13:52][D][voice_assistant:414]: State changed from IDLE to START_PIPELINE
[23:13:52][D][voice_assistant:420]: Desired state set to START_MICROPHONE
[23:13:52][D][voice_assistant:202]: Requesting start...
[23:13:52][D][voice_assistant:414]: State changed from START_PIPELINE to STARTING_PIPELINE
[23:13:52][D][voice_assistant:435]: Client started, streaming microphone
[23:13:52][D][voice_assistant:414]: State changed from STARTING_PIPELINE to STREAMING_MICROPHONE
[23:13:52][D][voice_assistant:420]: Desired state set to STREAMING_MICROPHONE
[23:13:52][D][voice_assistant:521]: Event Type: 1
[23:13:52][D][voice_assistant:524]: Assist Pipeline running
[23:13:52][D][voice_assistant:521]: Event Type: 9
[23:13:52][D][light:036]: 'top_led' Setting:
[23:13:52][D][light:051]:   Brightness: 60%
[23:13:52][D][light:059]:   Red: 100%, Green: 0%, Blue: 100%
[23:13:52][D][light:085]:   Transition length: 0.1s`

rosscullen, 2024-06-22:
Tried again with the new 2024.06 tweaks but no joy.... below is a summary log for anyone who can review and maybe point me in the right direction:

`INFO ESPHome 2024.6.1
INFO Reading configuration /config/esphome/onju-voice-1.yaml...
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-adf.git@v2.5
INFO Updating submodules (components/esp-adf-libs, components/esp-sr) for https://github.com/espressif/esp-adf.git@v2.5
INFO Updating https://github.com/espressif/esp-tflite-micro@None
INFO Compiling app...
Processing onju-voice-1 (board: esp32-s3-devkitc-1; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
/data/cache/platformio/packages/framework-espidf Already Patched: yes
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - framework-espidf @ 3.40407.240606 (4.4.7) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- Improv @ 1.2.3
|-- noise-c @ 0.1.4
Compiling .pioenvs/onju-voice-1/src/main.o
Linking .pioenvs/onju-voice-1/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: missing --end-group; added as last command line option
RAM:   [=         ]  11.2% (used 36792 bytes from 327680 bytes)
Flash: [========  ]  82.1% (used 1506053 bytes from 1835008 bytes)
Building .pioenvs/onju-voice-1/firmware.bin
Creating esp32s3 image...
Successfully created esp32s3 image.
esp32_create_combined_bin([".pioenvs/onju-voice-1/firmware.bin"], [".pioenvs/onju-voice-1/firmware.elf"])
Wrote 0x17fc70 bytes to file /data/build/onju-voice-1/.pioenvs/onju-voice-1/firmware.factory.bin, ready to flash to offset 0x0
esp32_copy_ota_bin([".pioenvs/onju-voice-1/firmware.bin"], [".pioenvs/onju-voice-1/firmware.elf"])
========================= [SUCCESS] Took 88.14 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.9.16
INFO Uploading /data/build/onju-voice-1/.pioenvs/onju-voice-1/firmware.bin (1506416 bytes)
Uploading: [============================================================] 100% Done...

INFO Upload took 12.92 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.9.16 using esphome API
INFO Successfully connected to onju-voice-1 @ 192.168.9.16 in 11.242s
INFO Successful handshake with onju-voice-1 @ 192.168.9.16 in 0.090s
[23:33:16][I][app:102]: ESPHome version 2024.2.0 compiled on Feb 25 2024, 12:33:07
[23:33:16][C][wifi:577]: WiFi:
[23:33:16][C][wifi:409]:   Local MAC: DC:DA:0C:2A:45:D8
[23:33:16][C][wifi:414]:   SSID: [redacted]
[23:33:16][C][wifi:415]:   IP Address: 192.168.9.16
[23:33:16][C][wifi:417]:   BSSID: [redacted]
[23:33:16][C][wifi:418]:   Hostname: 'onju-voice-1'
[23:33:16][C][wifi:420]:   Signal strength: -63 dB ▂▄▆█
[23:33:16][C][wifi:424]:   Channel: 6
[23:33:16][C][wifi:425]:   Subnet: 255.255.255.0
[23:33:16][C][wifi:426]:   Gateway: 192.168.9.1
[23:33:16][C][wifi:427]:   DNS1: 192.168.9.1
[23:33:16][C][wifi:428]:   DNS2: 0.0.0.0
[23:33:16][C][logger:447]: Logger:
[23:33:16][C][logger:448]:   Level: DEBUG
[23:33:16][C][logger:449]:   Log Baud Rate: 115200
[23:33:16][C][logger:451]:   Hardware UART: USB_CDC
[23:33:16][C][template.number:050]: Template Number 'Touch threshold percentage'
[23:33:16][C][template.number:051]:   Optimistic: YES
[23:33:16][C][template.number:052]:   Update Interval: never
[23:33:16][C][esp32_rmt_led_strip:175]: ESP32 RMT LED Strip:
[23:33:16][C][esp32_rmt_led_strip:176]:   Pin: 11
[23:33:16][C][esp32_rmt_led_strip:177]:   Channel: 0
[23:33:16][C][esp32_rmt_led_strip:202]:   RGB Order: GRB
[... 31 lines cut ...]
[23:33:17][C][esp32_touch:245]:   Denoise capacitance level: L0
[23:33:17][C][esp32_touch:260]:   Touch Pad 'volume_down'
[23:33:17][C][esp32_touch:261]:     Pad: T4
[23:33:17][C][esp32_touch:262]:     Threshold: 441320
[23:33:17][C][esp32_touch:260]:   Touch Pad 'volume_up'
[23:33:17][C][esp32_touch:261]:     Pad: T2
[23:33:17][C][esp32_touch:262]:     Threshold: 523781
[23:33:17][C][esp32_touch:260]:   Touch Pad 'action'
[23:33:17][C][esp32_touch:261]:     Pad: T3
[23:33:17][C][esp32_touch:262]:     Threshold: 627966
[23:33:17][C][captive_portal:088]: Captive Portal:
[23:33:17][C][mdns:115]: mDNS:
[23:33:17][C][mdns:116]:   Hostname: onju-voice-1
[23:33:17][C][ota:096]: Over-The-Air Updates:
[23:33:17][C][ota:097]:   Address: onju-voice-1.local:3232
[23:33:17][C][ota:100]:   Using Password.
[23:33:17][C][ota:103]:   OTA version: 2.
[23:33:17][C][api:139]: API Server:
[23:33:17][C][api:140]:   Address: onju-voice-1.local:6053
[23:33:17][C][api:142]:   Using noise encryption: YES
[23:33:17][C][audio:203]: Audio:
[23:33:17][C][audio:225]:   External DAC channels: 1
[23:33:17][C][audio:226]:   I2S DOUT Pin: 12`

Thanks

closed 2024-06-26 by rosscullen