tetele/onju-voice-satellite #60: Volume too loud or to silent with microwakeword

Mugga6315, 2024-07-05:
### Flavor

MicroWakeWord

### Checklist

- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/tetele/onju-voice-satellite/issues?q=is%3Aissue+label%3A%22bug%22+)..
- [X] I am able to consult debug logging for my ESPHome instance.

### Describe the issue

I just switched to mircowakeword for one satellite I'm using and the volume is quite high compared to non microwakeword built.
For example 50% volume sounds like 100% of non microwakeword variante. And going down to 35% is very silent.

Any idea what could be the issue here?

Using esphome 2024.6.6

### Reproduction steps

1. Issue command to satellite
2. Response very loud at 50% compared to non microwakeword
3.
...

tetele, 2024-07-12:
Thanks for reporting it, this has to do with the upstream [esphome_audio](https://github.com/gnumpi/esphome_audio) library, which already has [a branch ](https://github.com/gnumpi/esphome_audio/tree/alc_volume_curve) for this particular feature.

I've simply forgotten to add the missing parameter to the config.

tetele, 2024-07-12:
@Mugga6315 please confirm this is working as expected for you now

closed 2024-07-12 by tetele

Mugga6315, 2024-07-15:
@tetele Seems to be working now. Even tough I now think its a little bit more silent overall than the non microword version, but for me Its okay.

PR #65 (2024-07-12): Fix max volume on esphome_audio
Use `alc_max` parameter added in `esphome_audio` https://github.com/gnumpi/esphome_audio/tree/alc_volume_curve

Fixes #60