tetele/onju-voice-satellite #70: Timers!

jherby2k, 2024-07-30:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/tetele/onju-voice-satellite/issues?q=is%3Aissue+label%3A%22feature-request%22+).

### Is your feature request related to a problem? Please describe.

wish i could set timers

### Describe the solution you'd like

Timers are now supported in ESPHome/HASS but i think you need to add some code to make it work (right now Onju responds with "timers are not supported on this device).

Assume some or all of these need to be implemented:

on_timer_started (Optional, [Automation](https://esphome.io/automations/#automation)): An automation to perform when a voice assistant timer has started. The timer is available as timer of type [voice_assistant::Timer](https://esphome.io/api/structesphome_1_1voice__assistant_1_1_timer).

on_timer_finished (Optional, [Automation](https://esphome.io/automations/#automation)): An automation to perform when a voice assistant timer has finished. The timer is available as timer of type [voice_assistant::Timer](https://esphome.io/api/structesphome_1_1voice__assistant_1_1_timer).

on_timer_cancelled (Optional, [Automation](https://esphome.io/automations/#automation)): An automation to perform when a voice assistant timer has been cancelled. The timer is available as timer of type [voice_assistant::Timer](https://esphome.io/api/structesphome_1_1voice__assistant_1_1_timer).

on_timer_updated (Optional, [Automation](https://esphome.io/automations/#automation)): An automation to perform when a voice assistant timer has been updated (paused/resumed/duration changed). The timer is available as timer of type [voice_assistant::Timer](https://esphome.io/api/structesphome_1_1voice__assistant_1_1_timer).

on_timer_tick (Optional, [Automation](https://esphome.io/automations/#automation)): An automation to perform when the voice assistant timers tick is triggered. This is called every 1 second while there are timers on this device. The timers are available as timers which is a std::vector (array) of type [voice_assistant::Timer](https://esphome.io/api/structesphome_1_1voice__assistant_1_1_timer).

### Describe alternatives you've considered

N/A

### Additional context

Timers are already working on Atom Echo Dev Kit and the ESP32-S3 Box 3. Assume their code would be similar.

jhbruhn, 2024-07-31:
The problem with timers is that a sound needs to be played. The Onju Voice uses a media_player component instead of a speaker component though (compared to Atom Echo and S3 Box). While firmware-internal PCM Sounds can be played via the speaker component, the media_player component can, in its current state, only play remote files.

So we would either have to switch over to the speaker component and use the media_players remote file playback capabiltiies, or instruct all users to host an alert sound on an external server the media_player can play.

I hope that in the future, the media_player can also play back local sounds (see e.g. here https://github.com/gnumpi/esphome_audio/issues/56 or here https://github.com/esphome/voice-kit/tree/dev).

dreimer1986, 2024-08-03:
It's a lil nuissance, but it works fine. Most likely the assistant is fed by a online AI as local AI implementations and their support for HA have only appeared in the last few days. So you need a active internet connection anyways. Here is my lil fork where I tinker around a bit and sort of implemented timers successfully. Just go and steal of it what you want. Same goes for tetele of course ^^ I would be happy if my stuff is helpful for more than just me. As my stuff is from a absolute ESPHome beginner, I did not dare to open a PR with it as I had no one proof reading it yet.

P.S. please do not discuss a fork in the report though. This report is for the original code tree.

PR #74 (2024-08-07): Timers and chimes
Fix #70 

Implement timer support `voice_assistant`.

Also add some notification sounds for:
- wake word detected (MWW only, we can't pause the pipeline for non-MWW)
- ~~VAD ended~~ impossible because we can't pause the pipeline for chimes
- error
- timer finished
closed 2024-08-07 by tetele