tetele/onju-voice-satellite #78: Spotify Connect Support

lunDreame, 2024-08-10:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/tetele/onju-voice-satellite/issues?q=is%3Aissue+label%3A%22feature-request%22+).

### Is your feature request related to a problem? Please describe.

-

### Describe the solution you'd like

I've been thinking about turning it from the existing onju voice into spotify, but I couldn't find it. (Maybe I can't find it) I hope this onju voice can also be used for music sensibility. Any good alternatives?

### Describe alternatives you've considered

-

### Additional context

-

tetele, 2024-08-10:
ESPHome devices dpon't play Spotify without some middleware like Music Assistant

lunDreame, 2024-08-10:
> ESPHome devices dpon't play Spotify without some middleware like Music Assistant

It was installed, but I couldn’t because the entity didn‘t pop up. The entity just popped up!

closed 2024-08-10 by lunDreame