tetele/onju-voice-satellite #79: Chimes are quite slow from https

dreimer1986, 2024-08-10:
### Flavor

MicroWakeWord

### Checklist

- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/tetele/onju-voice-satellite/issues?q=is%3Aissue+label%3A%22bug%22+)..
- [X] I am able to consult debug logging for my ESPHome instance.

### Describe the issue

Current config loads the error, timer finished and wakeup chimes from github by https. The result is a quite long wait time until it plays the sound. I tried my luck with some tinkering, like:

- Adding CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240: "y"
- Commenting out codec: auto

But both did not really change anything at all. Then I had the idea that https is quite heavy, too and the S3 is not that of a super computer tbh. I move the three files to my HA www folder and changed the three variables to:

```
  wakeup_sound_url: "http://192.168.181.42:8123/local/wakeup.mp3"
  error_sound_url: "http://192.168.181.42:8123/local/error.mp3"
  timer_finished_sound_url: "http://192.168.181.42:8123/local/timer_finished.mp3"
```

The difference is from several seconds to wait down to almost instant playback. Of course you just could use homeassistant.local instead of the static IP here, but I chose to skip the DNS, too.

Here two videos. One with HTTPS and one with HTTP: https://photos.app.goo.gl/rSeHgzJCGFNnrnFU7

As long as we have no way to get the chimes into the firmware to run them locally, maybe a lil howto move the sounds to your HA instance as unencrypted host would be a good idea?

### Reproduction steps

Not much to do, get the current config on your Onju and start talking.

### Debug logs

```text
Local HTTP:
INFO ESPHome 2024.7.3
INFO Reading configuration /config/esphome/onju-voice-e27688.yaml...
INFO Starting log output from 192.168.181.103 using esphome API
INFO Successfully connected to onju-voice-e27688 @ 192.168.181.103 in 0.110s
INFO Successful handshake with onju-voice-e27688 @ 192.168.181.103 in 0.051s
[13:17:54][I][app:100]: ESPHome version 2024.7.3 compiled on Aug 10 2024, 07:53:16
[13:17:54][I][app:102]: Project tetele.onju_voice_satellite version 1.1.0
[13:17:54][C][wifi:599]: WiFi:
[13:17:54][C][wifi:427]:   Local MAC: 24:58:7C:E2:76:88
[13:17:54][C][wifi:432]:   SSID: 'Asgaard'[redacted]
[13:17:54][C][wifi:435]:   IP Address: 192.168.181.103
[13:17:54][C][wifi:439]:   BSSID: B0:F2:08:55:4B:1E[redacted]
[13:17:54][C][wifi:440]:   Hostname: 'onju-voice-e27688'
[13:17:54][C][wifi:442]:   Signal strength: -53 dB ▂▄▆█
[13:17:54][C][wifi:446]:   Channel: 13
[13:17:54][C][wifi:447]:   Subnet: 255.255.255.0
[13:17:54][C][wifi:448]:   Gateway: 192.168.181.254
[13:17:54][C][wifi:449]:   DNS1: 192.168.181.42
[13:17:54][C][wifi:450]:   DNS2: 0.0.0.0
[13:17:54][C][logger:185]: Logger:
[13:17:54][C][logger:186]:   Level: DEBUG
[13:17:54][C][logger:188]:   Log Baud Rate: 115200
[13:17:54][C][logger:189]:   Hardware UART: USB_SERIAL_JTAG
[13:17:54][C][template.number:050]: Template Number 'Touch threshold percentage'
[13:17:54][C][template.number:051]:   Optimistic: YES
[13:17:54][C][template.number:052]:   Update Interval: never
[13:17:54][C][esp32_rmt_led_strip:175]: ESP32 RMT LED Strip:
[13:17:54][C][esp32_rmt_led_strip:176]:   Pin: 11
[13:17:54][C][esp32_rmt_led_strip:177]:   Channel: 0
[13:17:54][C][esp32_rmt_led_strip:202]:   RGB Order: GRB
[13:17:54][C][esp32_rmt_led_strip:203]:   Max refresh rate: 0
[... 876 lines cut ...]
[13:28:19][I][adf_media_player:258]: play_intent: false
[13:28:19][I][adf_media_player:259]: current_uri_: yes
[13:28:20][D][esp-idf:000][decoder]: I (465829) AUDIO_ELEMENT: IN-[decoder] AEL_IO_DONE,-2

[13:28:20][D][esp-idf:000][decoder]: I (466169) MP3_DECODER: Closed

[13:28:20][I][esp_adf_pipeline:123]: [ decoder ] byte_pos: 0, total: -1276
[13:28:20][I][esp_adf_pipeline:132]: [ decoder ] status: 15
[13:28:20][I][esp_adf_pipeline:135]: current state: FINISHING
[13:28:20][D][esp-idf:000][resampler]: I (466275) AUDIO_ELEMENT: IN-[resampler] AEL_IO_DONE,-2

[13:28:20][I][esp_adf_pipeline:132]: [ resampler ] status: 15
[13:28:20][I][esp_adf_pipeline:135]: current state: FINISHING
[13:28:20][D][esp-idf:000][i2s_out]: I (466306) AUDIO_ELEMENT: IN-[i2s_out] AEL_IO_DONE,-2

[13:28:21][I][esp_adf_pipeline:123]: [ i2s_out ] byte_pos: 0, total: 0
[13:28:21][I][esp_adf_pipeline:132]: [ i2s_out ] status: 15
[13:28:21][I][esp_adf_pipeline:135]: current state: FINISHING
[13:28:21][D][esp_adf_pipeline:448]: [MediaPlayer] Pipeline changed from FINISHING to STOPPED. (REQ: 1)
[13:28:21][I][adf_media_player:192]: got new pipeline state: 4, while in MP state ANNOUNCING
[13:28:21][I][adf_media_player:256]: current mp state: IDLE
[13:28:21][I][adf_media_player:257]: anouncement: false
[13:28:21][I][adf_media_player:258]: play_intent: false
[13:28:21][I][adf_media_player:259]: current_uri_: yes
```

tetele, 2024-08-10:
Instructions are already in the readme, albeit with local hostname https://github.com/tetele/onju-voice-satellite?tab=readme-ov-file#notification-sounds 🤷🏻‍♂️

dreimer1986, 2024-08-10:
Oh. That's quite embarrassing. Sorry, did not see that phrase...

closed 2024-08-10 by dreimer1986