tetele/onju-voice-satellite #82: Adding a button in the HA panel that allows you to speak to the speaker without use wake words

witold-gren, 2024-09-14:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/tetele/onju-voice-satellite/issues?q=is%3Aissue+label%3A%22feature-request%22+).

### Is your feature request related to a problem? Please describe.

Is it possible to programmatically start listening for a response without using the wakewords? 

I would like to create some automation in HA which will say some text on the speaker (TTS), and then the speaker will start listening to respond. 

Example: At the specified time, the speaker will say: "Welcome home, can I help you with something?" - then I will be able to answer, "Turn on coffee machine". 

Example of automation:
```
alias: Onju Voice 
description: ""
trigger:
  - platform: time
    at: "10:00:00"
condition: []
action:
  - action: tts.speak
    metadata: {}
    data:
      cache: true
      media_player_entity_id: media_player.kuchnia_onju_voice_kuchnia_onju_voice
      message: >-
          It's 10:00 am. Can I do something for you?
      options:
        voice: wg_glos_meski
    target:
      entity_id: tts.piper
  - wait_for_trigger:
      - platform: state
        entity_id:
          - media_player.kuchnia_onju_voice_kuchnia_onju_voice
        from: playing
        to: idle
    timeout:
      hours: 0
      minutes: 0
      seconds: 20
      milliseconds: 0
    continue_on_timeout: false
  - action: esphome.kuchnia_onju_voice_start_va
    metadata: {}
    data: {}
    enabled: true
  - delay:
      hours: 0
      minutes: 0
      seconds: 5
      milliseconds: 0
  - action: esphome.kuchnia_onju_voice_stop_va
    metadata: {}
    data: {}
    enabled: true
mode: single
```

### Describe the solution you'd like

Adding a button to the HA panel, clicking it will start listening. Currently, after starting the `start_va` website, if we start talking and finish, the microphone is constantly listening (BTW. Unfortunately, it doesn't work properly for me). But I prefer to wake up the microphone and after I stop talking, I would like it to stop listening (just like it works with wake words).

### Describe alternatives you've considered

I try to use service:
```
[... 244 lines cut ...]
[15:43:24][D][esp-idf:000]: W (199638) AUDIO_EVT: There is no space in external queue
[15:43:24][D][esp-idf:000]: W (199642) AUDIO_ELEMENT: IN-[pcm_reader] AEL_IO_ABORT
[15:43:24][D][esp-idf:000]: W (199644) AUDIO_EVT: There is no space in external queue
[15:43:24][D][esp-idf:000]: W (199646) AUDIO_ELEMENT: IN-[pcm_reader] AEL_IO_ABORT
[15:43:24][D][esp-idf:000]: W (199649) AUDIO_EVT: There is no space in external queue
[15:43:24][D][esp-idf:000]: W (199653) AUDIO_ELEMENT: IN-[pcm_reader] AEL_IO_ABORT
[15:43:24][D][esp-idf:000]: W (199656) AUDIO_EVT: There is no space in external queue
[15:43:24][D][esp-idf:000]: W (199658) AUDIO_ELEMENT: IN-[pcm_reader] AEL_IO_ABORT
[15:43:24][D][esp-idf:000]: W (199660) AUDIO_EVT: There is no space in external queue
[15:43:24][D][esp-idf:000]: W (199663) AUDIO_ELEMENT: IN-[pcm_reader] AEL_IO_ABORT
[15:43:24][D][esp-idf:000]: W (199666) AUDIO_EVT: There is no space in external queue
[15:43:24][D][esp-idf:000]: W (199669) AUDIO_ELEMENT: IN-[pcm_reader] AEL_IO_ABORT
[15:43:24][D][esp-idf:000]: W (199672) AUDIO_EVT: There is no space in external queue
[15:43:24][D][esp-idf:000]: W (199674) AUDIO_ELEMENT: IN-[pcm_reader] AEL_IO_ABORT
[15:43:24][D][esp-idf:000]: W (199677) AUDIO_EVT: There is no space in external queue
[15:43:24][D][esp-idf:000]: W (199681) AUDIO_ELEMENT: IN-[pcm_reader] AEL_IO_ABORT
[15:43:24][D][esp-idf:000]: W (199684) AUDIO_EVT: There is no space in external queue
[15:43:24][D][esp-idf:000]: W (199686) AUDIO_ELEMENT: IN-[pcm_reader] AEL_IO_ABORT
[15:43:24][D][esp-idf:000]: W (199689) AUDIO_EVT: There is no space in external queue
...
```

### Additional context

I would like to create more interactive automations 😀at the moment you have to trigger the device every time to start listening.

tetele, 2024-09-14:
You can already do that (if i understand correctly). There are API services exposed to HA for starting and stopping the VA. The actual service name depends on your device name, but it should be along the lines of `esphome.onju_start_va`.

Let me know if i understood correctly or close the issue if this was what you were after.

witold-gren, 2024-09-14:
As I described above, this method unfortunately does not work as well as I would expect. When I turn on listening, I have to turn it off later. I would prefer that the listening turned off by itself. Another problem is that the indicated method does not work properly and after it is completed (`esphome.onju_stop_va`), the device restarts.

witold-gren, 2024-09-14:
All the time I see this logs when the speaker stop listening:
```
[18:25:57][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[18:25:57][D][esp32.preferences:114]: Saving 1 preferences to flash...
[18:25:57][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[18:26:16][D][voice_assistant:627]: Event Type: 11
[18:26:16][D][voice_assistant:783]: Starting STT by VAD
[18:26:20][D][voice_assistant:627]: Event Type: 12
[18:26:20][D][voice_assistant:787]: STT by VAD end
[18:26:20][D][voice_assistant:504]: State changed from STREAMING_MICROPHONE to STOP_MICROPHONE
[18:26:20][D][voice_assistant:510]: Desired state set to AWAITING_RESPONSE
[18:26:20][D][esp_adf_pipeline:070]: Called 'stop' while in RUNNING state.
[18:26:20][D][voice_assistant:504]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[18:26:20][D][esp_adf_pipeline:448]: [ADFMicrophone] Pipeline changed from RUNNING to ABORTING. (REQ: 1)
[18:26:20][D][light:036]: 'top_led' Setting:
[18:26:20][D][light:051]:   Brightness: 70%
[18:26:20][D][light:059]:   Red: 0%, Green: 20%, Blue: 100%
[18:26:20][D][light:109]:   Effect: 'processing'
[18:26:20][D][adf_audio_element:324]: [i2s_in] Checking State for stopping, got 3
[18:26:21][D][adf_audio_element:324]: [pcm_reader] Checking State for stopping, got 3
[18:26:21][D][esp-idf:000][i2s_in]: W (85058) AUDIO_ELEMENT: OUT-[i2s_in] AEL_IO_ABORT
[18:26:21][D][esp-idf:000][i2s_in]: W (85062) AUDIO_ELEMENT: OUT-[i2s_in] AEL_IO_ABORT
[18:26:21][D][esp-idf:000][i2s_in]: W (85066) AUDIO_ELEMENT: OUT-[i2s_in] AEL_IO_ABORT
[18:26:21][D][esp_adf_pipeline:448]: [ADFMicrophone] Pipeline changed from ABORTING to STOPPED. (REQ: 1)
[18:26:21][D][voice_assistant:504]: State changed from STOPPING_MICROPHONE to AWAITING_RESPONSE
[18:26:21][D][esp-idf:000]: W (85089) AUDIO_ELEMENT: IN-[pcm_reader] AEL_IO_ABORT
[18:26:21][D][esp-idf:000]: W (85092) AUDIO_ELEMENT: IN-[pcm_reader] AEL_IO_ABORT
[18:26:21][D][esp-idf:000]: W (85096) AUDIO_ELEMENT: IN-[pcm_reader] AEL_IO_ABORT
[18:26:21][D][esp-idf:000]: W (85098) AUDIO_ELEMENT: IN-[pcm_reader] AEL_IO_ABORT
[18:26:21][D][esp-idf:000]: W (85102) AUDIO_ELEMENT: IN-[pcm_reader] AEL_IO_ABORT
[18:26:21][D][esp-idf:000]: W (85105) AUDIO_EVT: There is no space in external queue
[18:26:21][D][esp-idf:000]: W (85108) AUDIO_ELEMENT: IN-[pcm_reader] AEL_IO_ABORT
[18:26:21][D][esp-idf:000]: W (85112) AUDIO_EVT: There is no space in external queue
[18:26:21][D][esp-idf:000]: W (85115) AUDIO_ELEMENT: IN-[pcm_reader] AEL_IO_ABORT
[18:26:21][D][esp-idf:000]: W (85118) AUDIO_EVT: There is no space in external queue
[18:26:21][D][esp-idf:000]: W (85120) AUDIO_ELEMENT: IN-[pcm_reader] AEL_IO_ABORT
[18:26:21][D][esp-idf:000]: W (85124) AUDIO_EVT: There is no space in external queue
[18:26:21][D][esp-idf:000]: W (85127) AUDIO_ELEMENT: IN-[pcm_reader] AEL_IO_ABORT
[18:26:21][D][esp-idf:000]: W (85130) AUDIO_EVT: There is no space in external queue
[18:26:21][D][esp-idf:000]: W (85133) AUDIO_ELEMENT: IN-[pcm_reader] AEL_IO_ABORT
[18:26:21][D][esp-idf:000]: W (85136) AUDIO_EVT: There is no space in external queue
[18:26:21][D][esp-idf:000]: W (85140) AUDIO_ELEMENT: IN-[pcm_reader] AEL_IO_ABORT
[18:26:21][D][esp-idf:000]: W (85143) AUDIO_EVT: There is no space in external queue
[18:26:21][D][esp-idf:000]: W (85145) AUDIO_ELEMENT: IN-[pcm_reader] AEL_IO_ABORT
[18:26:21][D][esp-idf:000]: W (85147) AUDIO_EVT: There is no space in external queue
[18:26:21][D][esp-idf:000]: W (85151) AUDIO_ELEMENT: IN-[pcm_reader] AEL_IO_ABORT
[18:26:21][D][esp-idf:000]: W (85153) AUDIO_EVT: There is no space in external queue
[18:26:21][D][esp-idf:000]: W (85157) AUDIO_ELEMENT: IN-[pcm_reader] AEL_IO_ABORT
[18:26:21][D][esp-idf:000]: W (85160) AUDIO_EVT: There is no space in external queue
[18:26:21][D][esp-idf:000]: W (85163) AUDIO_ELEMENT: IN-[pcm_reader] AEL_IO_ABORT
[18:26:21][D][esp-idf:000]: W (85166) AUDIO_EVT: There is no space in external queue
[18:26:21][D][esp-idf:000]: W (85168) AUDIO_ELEMENT: IN-[pcm_reader] AEL_IO_ABORT
[18:26:21][D][esp-idf:000]: W (85171) AUDIO_EVT: There is no space in external queue
[18:26:21][D][esp-idf:000]: W (85173) AUDIO_ELEMENT: IN-[pcm_reader] AEL_IO_ABORT
[18:26:21][D][esp-idf:000]: W (85175) AUDIO_EVT: There is no space in external queue
[18:26:21][D][esp-idf:000]: W (85178) AUDIO_ELEMENT: IN-[pcm_reader] AEL_IO_ABORT
[18:26:21][D][esp-idf:000]: W (85180) AUDIO_EVT: There is no space in external queue
[18:26:21][D][esp-idf:000]: W (85182) AUDIO_ELEMENT: IN-[pcm_reader] AEL_IO_ABORT
[18:26:21][D][esp-idf:000]: W (85186) AUDIO_EVT: There is no space in external queue
[18:26:21][D][esp-idf:000]: W (85189) AUDIO_ELEMENT: IN-[pcm_reader] AEL_IO_ABORT
[18:26:21][D][esp-idf:000]: W (85192) AUDIO_EVT: There is no space in external queue
[18:26:21][D][esp-idf:000]: W (85195) AUDIO_ELEMENT: IN-[pcm_reader] AEL_IO_ABORT
[18:26:21][D][esp-idf:000]: W (85198) AUDIO_EVT: There is no space in external queue
[18:26:21][D][esp-idf:000]: W (85202) AUDIO_ELEMENT: IN-[pcm_reader] AEL_IO_ABORT
[18:26:21][D][esp-idf:000]: W (85204) AUDIO_EVT: There is no space in external queue
[18:26:21][D][esp-idf:000]: W (85207) AUDIO_ELEMENT: IN-[pcm_reader] AEL_IO_ABORT
[18:26:21][D][esp-idf:000]: W (85210) AUDIO_EVT: There is no space in external queue
...
```

witold-gren, 2024-09-26:
Solution you can find here https://github.com/tetele/onju-voice-satellite/discussions/88

closed 2024-09-26 by witold-gren