tetele/onju-voice-satellite #85: Problem with long wait for response and sound effect

witold-gren, 2024-09-20:
### Flavor

MicroWakeWord

### Checklist

- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/tetele/onju-voice-satellite/issues?q=is%3Aissue+label%3A%22bug%22+)..
- [X] I am able to consult debug logging for my ESPHome instance.

### Describe the issue

Hi, I don't really understand why all the responses to my configuration are very slow. Every time I ask a question I have to wait a few seconds for an answer. You can see it very nicely in the video at the link: https://www.youtube.com/watch?v=TcIm_3co8XQ Is there anything I can do to make the sound effect trigger faster and the response to be generated faster?

[image]

[image]

[image]

I have **Home Assistance** installed on Proxmox running on an Intel NUC13ANHi7 13th Gen Core I7-1360P. I have no problem generating voices when I do it in the browser. You can also see in the photos that intent processing works really fast. Below is my configuration which I use in ESPHome:
```
substitutions:
  name: sypialnia-onju-voice
  friendly_name: Sypialnia Onju Voice
  project_version: "1.1.0"
  device_description: "Onju Voice Satellite with ESPHome software and microWakeWord"
  wakeup_sound_url: "https://homeassistant.local:8123/local/sounds/wakeup.mp3"
  error_sound_url: "https://homeassistant.local:8123/local/sounds/error.mp3"
  timer_finished_sound_url: "https://homeassistant.local:8123/local/sounds/timer_finished.mp3"

external_components:
  - source:
      type: git
      url: https://github.com/gnumpi/esphome_audio
      ref: dev-next
    components: [adf_pipeline, i2s_audio]

esphome:
  name: "${name}"
  friendly_name: "${friendly_name}"
  comment: "${device_description}"
  name_add_mac_suffix: false
  project:
    name: tetele.onju_voice_satellite
    version: "${project_version}"
  min_version: 2024.7.3
  platformio_options:
    board_build.flash_mode: dio
    board_build.arduino.memory_type: qio_opi
  on_boot:
    then:
      - light.turn_on:
          id: top_led
          effect: slow_pulse
          red: 100%
          green: 60%
          blue: 0%
      - wait_until:
          condition:
            wifi.connected:
      - light.turn_on:
          id: top_led
          effect: pulse
          red: 0%
          green: 100%
          blue: 0%
      - wait_until:
          condition:
            api.connected:
[... 1194 lines cut ...]
[16:42:31][I][adf_media_player:258]: play_intent: false
[16:42:31][I][adf_media_player:259]: current_uri_: yes
[16:42:31][D][esp-idf:000][decoder]: I (269820931) AUDIO_ELEMENT: IN-[decoder] AEL_IO_DONE,-2

[16:42:31][D][esp-idf:000][decoder]: I (269821021) MP3_DECODER: Closed

[16:42:31][I][esp_adf_pipeline:123]: [ decoder ] byte_pos: 0, total: -93
[16:42:31][I][esp_adf_pipeline:132]: [ decoder ] status: 15
[16:42:31][I][esp_adf_pipeline:135]: current state: FINISHING
[16:42:31][D][esp-idf:000][resampler]: I (269821052) AUDIO_ELEMENT: IN-[resampler] AEL_IO_DONE,-2

[16:42:31][I][esp_adf_pipeline:132]: [ resampler ] status: 15
[16:42:31][I][esp_adf_pipeline:135]: current state: FINISHING
[16:42:31][D][esp-idf:000][i2s_out]: I (269821131) AUDIO_ELEMENT: IN-[i2s_out] AEL_IO_DONE,-2

[16:42:32][I][esp_adf_pipeline:123]: [ i2s_out ] byte_pos: 0, total: 0
[16:42:32][I][esp_adf_pipeline:132]: [ i2s_out ] status: 15
[16:42:32][I][esp_adf_pipeline:135]: current state: FINISHING
[16:42:32][D][esp_adf_pipeline:448]: [MediaPlayer] Pipeline changed from FINISHING to STOPPED. (REQ: 1)
[16:42:32][I][adf_media_player:192]: got new pipeline state: 4, while in MP state PLAYING
[16:42:32][I][adf_media_player:256]: current mp state: IDLE
[16:42:32][I][adf_media_player:257]: anouncement: false
[16:42:32][I][adf_media_player:258]: play_intent: false
[16:42:32][I][adf_media_player:259]: current_uri_: yes
```

TheStigh, 2024-09-20:
1. You should **not** put the wakeup sound behind https, but http, AND, avoid DNS lookup and use IP as it do steal you for some milliseconds of lookup time. Do this and it will be immediate wakeup response!

2. Regarding waiting for the response - that is a combination - first is back to https but here you can't do anything as the firmware of the ESP32 is pointing to your external URL (really bad from the HA Team!) - it should at least pointed to the internal team. Https and downloading mp3 is not a really good combination.

Second part - I'm about to test something this weekend - hopefully this will improve the response time.

witold-gren, 2024-09-20:
@TheStigh Thanks for your answer, I will try to check your recommendation. 
PS. My DNS name of HA is completely different but I had to hide it so that the domain was not publicly available :)

witold-gren, 2024-09-20:
I can confirm, that when I replace https to http its start working really fast. Also I removed https connection from internal path and I replace it to don't use domain name but to use directly internal IP, and now piper response also really fast.
Home Assistant configuration file:
```
homeassistant:
  name: Mieszkanie
  ...
  external_url: "https://MY_EXTERNAL_DOMAIN"
  internal_url: "http://10.20.1.4:8123"
...
http:
  server_port: 8123
  use_x_forwarded_for: true
  trusted_proxies:
    - 127.0.0.1
    - 10.20.0.0/23
  # IMPORTANT: Removed this ssl certificates
  # ssl_certificate: /ssl/fullchain.pem
  # ssl_key: /ssl/privkey.pem
  ip_ban_enabled: true
  login_attempts_threshold: 10
```

Configuration of Onju Voice:
```
substitutions:
  name: salon-onju-voice
  friendly_name: Salon Onju Voice
  project_version: "1.1.0"
  device_description: "Onju Voice Satellite with ESPHome software and microWakeWord"
  wakeup_sound_url: "http://10.20.1.4:8123/local/sounds/wakeup.mp3"
  error_sound_url: "http://10.20.1.4:8123/local/sounds/error.mp3"
  timer_finished_sound_url: "http://10.20.1.4:8123/local/sounds/timer_finished.mp3"
...
```

Below you can find logs from my Onju Voice device:
```
INFO ESPHome 2024.9.0
INFO Reading configuration /config/esphome/kuchnia-onju-voice.yaml...
INFO Starting log output from 10.20.0.77 using esphome API
INFO Successfully connected to kuchnia-onju-voice @ 10.20.0.77 in 0.192s
INFO Successful handshake with kuchnia-onju-voice @ 10.20.0.77 in 0.072s
[21:35:15][I][app:100]: ESPHome version 2024.9.0 compiled on Sep 20 2024, 20:47:51
[21:35:15][I][app:102]: Project tetele.onju_voice_satellite version 1.1.0
[21:35:15][C][wifi:600]: WiFi:
[21:35:15][C][wifi:428]:   Local MAC: 64:E8:33:47:7A:98
[21:35:15][C][wifi:433]:   SSID: [redacted]
[21:35:15][C][wifi:436]:   IP Address: 10.20.0.77
[21:35:15][C][wifi:440]:   BSSID: [redacted]
[21:35:15][C][wifi:441]:   Hostname: 'kuchnia-onju-voice'
[21:35:15][C][wifi:443]:   Signal strength: -67 dB ▂▄▆█
[21:35:15][C][wifi:447]:   Channel: 4
[21:35:15][C][wifi:448]:   Subnet: 255.255.254.0
[21:35:15][C][wifi:449]:   Gateway: 10.20.0.1
[21:35:15][C][wifi:450]:   DNS1: 10.20.0.1
[21:35:15][C][wifi:451]:   DNS2: 0.0.0.0
[21:35:15][C][logger:185]: Logger:
[21:35:15][C][logger:186]:   Level: DEBUG
[21:35:15][C][logger:188]:   Log Baud Rate: 115200
[21:35:15][C][logger:189]:   Hardware UART: USB_SERIAL_JTAG
[21:35:15][C][template.number:050]: Template Number 'Touch threshold percentage'
[21:35:15][C][template.number:051]:   Optimistic: YES
[21:35:15][C][template.number:052]:   Update Interval: never
[21:35:15][C][esp32_rmt_led_strip:187]: ESP32 RMT LED Strip:
[21:35:15][C][esp32_rmt_led_strip:188]:   Pin: 11
[21:35:15][C][esp32_rmt_led_strip:189]:   Channel: 0
[21:35:15][C][esp32_rmt_led_strip:214]:   RGB Order: GRB
[21:35:15][C][esp32_rmt_led_strip:215]:   Max refresh rate: 0
[21:35:15][C][esp32_rmt_led_strip:216]:   Number of LEDs: 6
[21:35:15][C][switch.gpio:068]: GPIO Switch 'dac_mute'
[... 389 lines cut ...]
[21:35:33][D][esp-idf:000][decoder]: I (1921996) AUDIO_ELEMENT: IN-[decoder] AEL_IO_DONE,-2

[21:35:34][D][esp-idf:000][decoder]: I (1922440) MP3_DECODER: Closed

[21:35:34][I][esp_adf_pipeline:123]: [ decoder ] byte_pos: 0, total: -1147
[21:35:34][I][esp_adf_pipeline:132]: [ decoder ] status: 15
[21:35:34][I][esp_adf_pipeline:135]: current state: FINISHING
[21:35:34][D][esp-idf:000][resampler]: I (1922568) AUDIO_ELEMENT: IN-[resampler] AEL_IO_DONE,-2

[21:35:34][I][esp_adf_pipeline:132]: [ resampler ] status: 15
[21:35:34][I][esp_adf_pipeline:135]: current state: FINISHING
[21:35:34][D][esp-idf:000][i2s_out]: I (1922631) AUDIO_ELEMENT: IN-[i2s_out] AEL_IO_DONE,-2

[21:35:34][I][esp_adf_pipeline:123]: [ i2s_out ] byte_pos: 0, total: 0
[21:35:34][I][esp_adf_pipeline:132]: [ i2s_out ] status: 15
[21:35:34][I][esp_adf_pipeline:135]: current state: FINISHING
[21:35:34][D][esp_adf_pipeline:448]: [MediaPlayer] Pipeline changed from FINISHING to STOPPED. (REQ: 1)
[21:35:34][I][adf_media_player:192]: got new pipeline state: 4, while in MP state ANNOUNCING
[21:35:34][I][adf_media_player:256]: current mp state: IDLE
[21:35:34][I][adf_media_player:257]: anouncement: false
[21:35:34][I][adf_media_player:258]: play_intent: false
[21:35:34][I][adf_media_player:259]: current_uri_: yes
[21:35:36][D][voice_assistant:514]: State changed from STREAMING_RESPONSE to IDLE
[21:35:36][D][voice_assistant:520]: Desired state set to IDLE
```

witold-gren, 2024-09-20:
@TheStigh I have one more question. Currently, it works so quickly that the prefix `bo` is added to each sentence - it is definitely a translation of the signal. Can I somehow delay listening to the command by 1 second? 😀
[image]
Currently we have: "bo zgaś światło w sypialni" but it should be: "zgaś światło w sypialni" 😀 I see this prefix `bo` really in all sentences..

witold-gren, 2024-09-21:
I found a solution.. I just add `delay: 500ms` when wake word detected:
```
micro_wake_word:
  models:
    #- model: https://github.com/kahrendt/microWakeWord/releases/download/v2.1_models/alexa.json
    # - model: https://github.com/kahrendt/microWakeWord/releases/download/v2.1_models/okay_nabu.json
    - model: https://github.com/kahrendt/microWakeWord/releases/download/v2.1_models/hey_jarvis.json
    #- model: https://github.com/kahrendt/microWakeWord/releases/download/v2.1_models/hey_mycroft.json
  vad:
    model: https://github.com/kahrendt/microWakeWord/releases/download/v2.1_models/vad.json
  on_wake_word_detected:
    - if:
        condition: media_player.is_playing
        then:
          - media_player.pause
    - media_player.play_media: "${wakeup_sound_url}"
    - delay: 500ms
    - wait_until:
        not:
          media_player.is_playing: onju_out
    - voice_assistant.start:
        wake_word: !lambda return wake_word;
```

witold-gren, 2024-09-21:
All problems have been solved 😀 I think it is worth adding such information to the README.md file:
1. If possible, use an IP address instead of a domain name for an internal address
2. in internal network, do not use `https` connection, only `http` (encrypted connection delays communication very much)
cc: @tetele

closed 2024-09-21 by witold-gren

TheStigh, 2024-09-21:
> All problems have been solved 😀 I think it is worth adding such information to the README.md file:
>
> 1. If possible, use an IP address instead of a domain name for an internal address
> 2. in internal [...]

Great that it worked out :)
Though, how many seconds does it take from you've finished talking until you hear the sound? Here, I think it is more time to cut as we do download the response as an mp3 file, and this is done through also the external URL which is https (as you still use).

I *think* I've found a way to cut with at least 1 second but I will need help from somebody better with the ESPHome code than me.

witold-gren, 2024-09-21:
I post new video with example how fast it works.. https://youtu.be/h63s-1HTkN8?feature=shared 

Unfortunately, now due to the addition of a 500ms delay, the LED lighting works a bit strange - you can see it in the video. Unfortunately, I don't know how to properly deal with the listening delay. So far I haven't found any reasonable solution to deal with it..