tetele/onju-voice-satellite #86: Microphone not working on 4 boards

robsonke, 2024-09-21:
### Flavor

MicroWakeWord

### Checklist

- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/tetele/onju-voice-satellite/issues?q=is%3Aissue+label%3A%22bug%22+)..
- [X] I am able to consult debug logging for my ESPHome instance.

### Describe the issue

It's most likely not related to the esphome config of onju voice but I can really use some help and hints. I ordered 5 boards, so far I've tried 4 of them. Flashed them, did the hardware switch, checked the yt movies 10 times.

Buttons work, speaker works, leds work, home assistant integration works but the microphone seems broken. When I use a wake word, nothing responds, leds remain the same and I dont see anything in the logs. Whenever I tap the top and initiate listening (I see that in the logs,) but nothing is received and it times out.

For me it's hard to believe there's a hardware issue in 4 boards and both microphones. Did I make some silly mistake or how can I test this?

Thanks in advance, if it's inappropriate filing an issue, let me know.

### Reproduction steps

1. Say "Ok Nabu"
2. Wait
3. Or test without wake word, no sound received

TheStigh, 2024-09-25:
Did you take the file and flashed without any changes? And what version? What you should do is scale down to debug - test both the openWakeWord and microWakeWord edition. I hardly believe you've received 4 boards from PCBWay with defective mics (but who knows)

robsonke, 2024-09-25:
Thanks. No, I compared my yaml with the original one, it's really only minor changes such as the webserver and network info. I've enabled debug logging too but didn't give extra info apart from the lack of voice.

dreimer1986, 2024-09-25:
I suggest to use the official ESPHome firmware and tools as a test here. https://github.com/justLV/onju-voice if this fails, too then your only step helping here is reporting faulty boards over at PCBWay. I got my 5 boards from them, too and the only thing they are really bad at in my case are the LEDs. Looks like they don't survive the solder bath. I had to fix about 50% of them. Some did not work on one or two colors, some were completely dead. But the microphone never was a flaw here. Btw, the mic used is the one from the Google hardware. You sure that the ribbon cable is really sitting correctly?

robsonke, 2024-09-26:
I will try the original yaml when I'm back home, though I dont think it will make much difference. About the mic, that's not the Google hardware, it's the 2 small holes in the print on both sides. The Google part is just the speaker.

jherby2k, 2024-09-26:
I had a weird mic problem that never got satisfactorily solved. It seems the issue was with the actual nest hardware - i swapped the same board to a different speaker housing and it worked fine. Put another board into the first speaker housing and it also didn't work. 

Couldn't for the life of me figure it out. Tried trimming the foil back, looked at the working and not working ones with a magnifying glass. Nothing! Gave up in frustration and binned the nest.

dreimer1986, 2024-09-28:
Looks like you are right. There a parts with a M on both sides. Did not expect that the mics are that good and instead really thought the ribbon cable has enough contacts for a mic. Ok. Still I have none of these problems and really look forward to the test results.

robsonke, 2024-09-30:
I just installed the yaml without the wake word. Only had to set a fixed ip and an api encryption key. It seems to a bit further but I doubt if it really notices my voice. I tested with "can you turn off the office lights" and then you get:
Speech recognised as: " The Earth is the Earth of the Earth."

A part of the logs:
```
[20:43:10][D][voice_assistant:797]: STT by VAD end
[20:43:10][D][voice_assistant:514]: State changed from STREAMING_MICROPHONE to STOP_MICROPHONE
[20:43:10][D][voice_assistant:520]: Desired state set to AWAITING_RESPONSE
[20:43:10][D][voice_assistant:514]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[20:43:10][D][light:036]: 'top_led' Setting:
[20:43:10][D][light:051]:   Brightness: 70%
[20:43:10][D][light:059]:   Red: 0%, Green: 20%, Blue: 100%
[20:43:10][D][light:109]:   Effect: 'processing'
[20:43:10][D][voice_assistant:514]: State changed from STOPPING_MICROPHONE to AWAITING_RESPONSE
[20:43:13][D][voice_assistant:637]: Event Type: 4
[20:43:13][D][voice_assistant:665]: Speech recognised as: " The Earth is the Earth of the Earth."
[20:43:13][D][voice_assistant:637]: Event Type: 5
[20:43:13][D][voice_assistant:670]: Intent started
[20:43:13][D][voice_assistant:637]: Event Type: 6
[20:43:13][D][voice_assistant:637]: Event Type: 7
[20:43:13][D][voice_assistant:693]: Response: "Sorry, I am not aware of any device called earth is the earth of the earth."
```

dreimer1986, 2024-09-30:
Some testing firmware which records and plays back a sample would be nice for such cases... But everything sounds like the next step is reporting back to PCBWay and ask for support there...

robsonke, 2024-10-01:
I will do that, lets make it their issue. Thanks all so far.