tetele/onju-voice-satellite #92: can not build on esphome 2024-10

lecrams, 2024-10-21:
### Flavor

OpenWakeWord or no wake word

### Checklist

- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/tetele/onju-voice-satellite/issues?q=is%3Aissue+label%3A%22bug%22+)..
- [X] I am able to consult debug logging for my ESPHome instance.

### Describe the issue

When I try to build the configuration from the HA integration , the build is teminated (see logs below), Doing a clean build first gives the same results

### Reproduction steps

1. Install .yaml
2.
3.
...

### Debug logs

```text
INFO ESPHome 2024.10.0
INFO Reading configuration /config/esphome/onju-voice-2.yaml...
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-adf.git@v2.5
INFO Updating submodules (components/esp-adf-libs, components/esp-sr) for https://github.com/espressif/esp-adf.git@v2.5
INFO Updating https://github.com/espressif/esp-tflite-micro@v1.3.1
INFO Updating https://github.com/espressif/esp-nn@v1.1.0
INFO Compiling app...
Processing onju-voice-2 (board: esp32-s3-devkitc-1; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
/data/cache/platformio/packages/framework-espidf Already Patched: yes
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - framework-espidf @ 3.40408.0 (4.4.8) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- ArduinoJson @ 6.18.5
|-- Improv @ 1.2.4
|-- ESPMicroSpeechFeatures @ 1.1.0
Compiling .pioenvs/onju-voice-2/src/main.o
Compiling .pioenvs/onju-voice-2/components/esp-tflite-micro/tensorflow/lite/micro/fake_micro_context.o
Compiling .pioenvs/onju-voice-2/components/esp-tflite-micro/tensorflow/lite/micro/flatbuffer_utils.o
Compiling .pioenvs/onju-voice-2/components/esp-tflite-micro/tensorflow/lite/micro/micro_interpreter_graph.o
Compiling .pioenvs/onju-voice-2/components/esp-tflite-micro/tensorflow/lite/micro/micro_log.o
Compiling .pioenvs/onju-voice-2/components/esp-tflite-micro/tensorflow/lite/micro/micro_op_resolver.o
Compiling .pioenvs/onju-voice-2/components/esp-tflite-micro/tensorflow/lite/micro/micro_profiler.o
Compiling .pioenvs/onju-voice-2/components/esp-tflite-micro/tensorflow/lite/micro/micro_resource_variable.o
xtensa-esp32s3-elf-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
*** [.pioenvs/onju-voice-2/src/main.o] Error 1
xtensa-esp32s3-elf-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
*** [.pioenvs/onju-voice-2/components/esp-tflite-micro/tensorflow/lite/micro/fake_micro_context.o] Error 1
xtensa-esp32s3-elf-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
*** [.pioenvs/onju-voice-2/components/esp-tflite-micro/tensorflow/lite/micro/micro_interpreter_graph.o] Error 1
```

tetele, 2024-10-21:
This does not appear to have anything to do with the config. Can you compile any other config using Micro Wake Word on that machine? Can you compile the same Onju config on another machine?

lecrams, 2024-10-21:
When I try to build my S3 box with the HA integration I get the same error. Building from the windows command line gives me NO error (both onju and S3B.
I could of course build everything from the command line but I prefer the integration, so it would be great if we can find the root cause for this issue

tetele, 2024-10-21:
Right, but this is not the place for that :) Check out the [ESPHome Discord](https://discord.gg/mNU5gtxW) server.

closed 2024-10-21 by tetele

lecrams, 2024-10-21:
Okay then I'll open an issue there