tetele/onju-voice-satellite #102: Error during installation

lunDreame, 2024-12-22:
### Flavor

MicroWakeWord

### Checklist

- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/tetele/onju-voice-satellite/issues?q=is%3Aissue+label%3A%22bug%22+)..
- [X] I am able to consult debug logging for my ESPHome instance.

### Describe the issue

https://github.com/tetele/onju-voice-satellite/blob/main/esphome/onju-voice-microwakeword.yaml,
I got the yaml file from 

wifi:
ssid: !secret wifi_ssid
password: !secret wifi_password
# Set up a wifi access point
ap:
ssid: "${friendly_name}"

api:
encryption:
key: "DELETED"
services:
- service: start_va
then:
- voice_assistant.start
- service: stop_va
then:
- voice_assistant.stop
- service: notification_on
then:
- script.execute: turn_on_notification
- service: notification_clear
then:
- script.execute: clear_notification

I modified the wifi and api sections and saved them, but I got an error during installation. I need help.

### Reproduction steps

1.
2.
3.
...

### Debug logs

```text
INFO ESPHome 2024.12.2
INFO Reading configuration /config/esphome/onju-voice-54ae60.yaml...
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-adf.git@v2.5
INFO Updating submodules (components/esp-adf-libs, components/esp-sr) for https://github.com/espressif/esp-adf.git@v2.5
INFO Updating https://github.com/espressif/esp-tflite-micro@v1.3.1
INFO Updating https://github.com/espressif/esp-nn@v1.1.0
INFO Compiling app...
Processing onju-voice (board: esp32-s3-devkitc-1; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32.git#51.03.07)
--------------------------------------------------------------------------------
Library Manager: Installing improv/Improv @ 1.2.4
INFO Installing improv/Improv @ 1.2.4
Unpacking  [####################################]  100%
Library Manager: Improv@1.2.4 has been installed!
INFO Improv@1.2.4 has been installed!
Library Manager: Installing esphome/noise-c @ 0.1.6
INFO Installing esphome/noise-c @ 0.1.6
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.6 has been installed!
INFO noise-c@0.1.6 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.4
INFO Installing esphome/libsodium @ 1.10018.4
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.4 has been installed!
INFO libsodium@1.10018.4 has been installed!
Library Manager: Installing kahrendt/ESPMicroSpeechFeatures @ 1.1.0
INFO Installing kahrendt/ESPMicroSpeechFeatures @ 1.1.0
Unpacking  [####################################]  100%
Library Manager: ESPMicroSpeechFeatures@1.1.0 has been installed!
INFO ESPMicroSpeechFeatures@1.1.0 has been installed!
/data/cache/platformio/packages/framework-espidf Already Patched: yes
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - framework-espidf @ 3.50105.0 (5.1.5) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.8.1 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 12.2.0+20230208 
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208
Reading CMake configuration...
-- Found Git: /usr/bin/git (found version "2.39.5") 
-- git rev-parse returned 'fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).'
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
Processing 1 dependencies:
[1/1] idf (5.1.5)
-- Configuring incomplete, errors occurred!
See also "/data/build/onju-voice/.pioenvs/onju-voice/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:266 (message):
  Failed to resolve component 'esp_adc_cal'.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:302 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:309 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:603 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:604 (idf_build_process)
  CMakeLists.txt:3 (project)

========================== [FAILED] Took 5.05 seconds ==========================
```

thinkjk, 2024-12-22:
Hi, I believe this will be fixed if you follow this: https://github.com/tetele/onju-voice-satellite/issues/101#issuecomment-2551945331

lunDreame, 2024-12-23:
Oh, thanks I will tried

closed 2024-12-23 by lunDreame