tetele/onju-voice-satellite #115: can not build with esphome 2025.3.3

lecrams, 2025-04-03:
### Flavor

OpenWakeWord or no wake word

### Checklist

- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of currently [previous issues](https://github.com/tetele/onju-voice-satellite/issues?q=is%3Aissue+label%3A%22bug%22+)..
- [x] I am able to consult debug logging for my ESPHome instance.

### Describe the issue

can not build onju-voice with esphome 2025.3.3

### Reproduction steps

1.
2.
3.
...

### Debug logs

```text
C:\Users\marce\.platformio\packages\framework-espidf@src-6e6cd5244a43bdfe453daac50cfe73b6 Already Patched: yes
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (51.3.6+sha.ec17486) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf @ 3.50105.0 (5.1.5)
 - tool-cmake @ 3.21.3
 - tool-esptoolpy @ 4.8.1
 - tool-mklittlefs @ 3.2.0
 - tool-ninja @ 1.9.0
 - tool-riscv32-esp-elf-gdb @ 11.2.0+20220823
 - tool-xtensa-esp-elf-gdb @ 11.2.0+20230208
 - toolchain-esp32ulp @ 2.35.0-20220830
 - toolchain-riscv32-esp @ 12.2.0+20230208
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208
Reading CMake configuration...
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Building ESP-IDF components for target esp32s3
NOTICE:

Following dependencies have new versions available:

Dependency "espressif/esp-nn": "1.1.0" -> "1.1.1"

Consider running "idf.py update-dependencies" to update your lock file.

Processing 1 dependencies:

[1/1] idf (5.1.5)

-- Configuring incomplete, errors occurred!
See also "G:/esphome/.esphome/build/onju-voice-1/.pioenvs/onju-voice-1/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any of the parent directories): .git
CMake Error at C:/Users/marce/.platformio/packages/framework-espidf@src-6e6cd5244a43bdfe453daac50cfe73b6/tools/cmake/build.cmake:266 (message):
  Failed to resolve component 'esp_adc_cal'.
Call Stack (most recent call first):
  C:/Users/marce/.platformio/packages/framework-espidf@src-6e6cd5244a43bdfe453daac50cfe73b6/tools/cmake/build.cmake:302 (__build_resolve_and_add_req)
  C:/Users/marce/.platformio/packages/framework-espidf@src-6e6cd5244a43bdfe453daac50cfe73b6/tools/cmake/build.cmake:309 (__build_expand_requirements)
  C:/Users/marce/.platformio/packages/framework-espidf@src-6e6cd5244a43bdfe453daac50cfe73b6/tools/cmake/build.cmake:603 (__build_expand_requirements)
  C:/Users/marce/.platformio/packages/framework-espidf@src-6e6cd5244a43bdfe453daac50cfe73b6/tools/cmake/project.cmake:604 (idf_build_process)
  CMakeLists.txt:3 (project)

============================================== [FAILED] Took 3.51 seconds ==============================================
```

srwalter, 2025-07-24:
FYI I have a version that works with latest esphome (2025.7.3 as of writing)

https://github.com/srwalter/onju-voice-satellite/blob/fix-microwakeword/esphome/onju-voice-mww-new.yaml

srwalter, 2025-07-25:
Can you tell me exactly what version you're using to build?

ther3zz, 2025-07-25:
> I didn't hit that exact error, but I did find and fix another problem with 2025.7.3. Here's how I am building, which succeeds:
> 
> docker run -v $PWD:$PWD ghcr.io/esphome/esphome:2025.7.3 compile $PWD/esphome/onju-voice-mww-new.yaml

wait... where are you adding that? in the onjuvoice yaml file in home assistant?

srwalter, 2025-07-25:
I've built it 2 ways: One from Linux command line (that's the docker command above)

The other is "ESPHome Builder" in Home Assistant, in which case I just pasted all of onju-voice-mww-new.yaml into the editor.  Both of those configurations work for me

harmfulmushroom, 2025-07-28:
> > Can you tell me exactly what version you're using to build?
> 
> ESPHome 2025.7.3
> 
> Running it like:
> 
> ```
> packages:
>     esphome.voice-assistant: github://srwalter/onju-voice-satellite/esphome/onju-voice-mww-new.yaml@fix-microwakeword
> ```

Clean your build files. I had similar errors until I tried this on my end, worked flawlessly afterwards.

MTrab, 2025-09-08:
Seeing the same as @Helvio88
Anyone managed to fix this?